gotthardp/python-mercuryapi #2: Problem installing

karimi68, 2016-08-02:
When I run "sudo python3 setup.py install" I end up with the following error:

error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

How can I resolve the problem? (Tnx)

gotthardp, 2016-08-03:
I am not sure. Could you please post here the messages that are printed before? Everything that is printed on the screen after you run "sudo python3 setup.py install"?

gotthardp, 2016-08-13:
Hi again. Does the problem persists?

dkowalsky7, 2016-08-25:
I'm having problems installing the dependencies, which I'm assuming is the problem here. I am trying use this on ubuntu and I cannot enable the repos with yum. Is there anyway to get the dependencies with sudo apt-get? Is there anyway to get the package with windows, if its not a problem

gotthardp, 2016-08-25:
Something like `apt-get install gcc patch libxslt` should work as well. Will you please try it and tell me what happened? I have no way to test this.

dkowalsky7, 2016-08-25:
I get

 E: unable to locate package libxslt

gotthardp, 2016-08-25:
I see. Then please try `apt-get install gcc patch xsltproc`.

dkowalsky7, 2016-08-25:
I was able to install that package, however I get an error from the make command at the end
[image]

and then a get an error with setup.py

[image]

gotthardp, 2016-08-26:
Thank you very much for your patience. We are getting there slowly. There was one prerequisite missing, which is clearly my fault.

Please try `apt-get install patch xsltproc gcc libreadline-dev` and make the project again.

dkowalsky7, 2016-08-28:
worked great! thanks for the help

gotthardp, 2016-08-28:
I'm glad it did. Should you need anything else, just let me know.

closed 2016-08-28 by gotthardp