gotthardp/python-mercuryapi #3: Setting Device Power and Read Mode

ENAML, 2016-09-27:
Hi,

Thanks for making this library available. I'm trying to integrate into a project I'm working on, and I was wondering if there was any way to set a device's power and read mode in Python.

Thanks again.

gotthardp, 2016-09-28:
Hello. The python wrapper currently implements only a very small portion of the mercuryAPI because I didn't want to implement functions that nobody needs. There is currently no way to set the device's power and read mode, but I can easily implement those for you.

Could you please describe a bit more what exactly you need?
 * You want to set `/reader/radio/readPower` or `/reader/powerMode`?
 * You want to set pseudo-acynchronous or continuous reading, or you meant something different?

eheaton, 2016-09-28:
Hi @gotthardp - Thanks for the reply. To answer your questions.

- We're looking to set the `/radio/read/power`.
- Correct. We're looking to set the pseudo-async, continuous reading

gotthardp, 2016-09-28:
Hi @eheaton, few more questions please:
 * You mean `/reader/radio/readPower`, right? Will an single integer parameter in `set_read_plan` do the job you need? E.g. `reader.set_read_plan([1], "GEN2", power=10)`? 
 * Do you have any thoughts on how the async-read should be architected? How shall the main loop look like? What makes the synch read() not suitable for your project?

eheaton, 2016-09-28:
Sorry, yes - I meant  `/reader/radio/readPower` ... the single int would be perfect.

As far as the sync read method, we're looking for a way to keep the reader in a continuous state to always be looking for tags once they get close. We currently have it looping over the read function every X seconds, which isn't bad, but we were under the impression that there's a mode you can put the reader in to do this natively.

ENAML, 2016-09-28:
Hey @gotthardp,

Thanks for adding this functionality! I've been trying to get your updates working, but I'm encountering what I believe to be an error in how the "set_read_plan" method in Python is passing arguments to C. See below:

```
Traceback (most recent call last):
  File "read.py", line 12, in <module>
    reader.set_read_plan([1], "GEN2", 1900)
RuntimeError: More keyword list entries (3) than format specifiers (2)
```

Would you mind taking a look at this? I've tried recompiling after making tweaks to "Reader_set_read_plan" in C but nothing seems to have worked. Thanks again!

gotthardp, 2016-09-28:
you should specify the keyword, i.e. do `reader.set_read_plan([1], "GEN2", read_power=1900)`

ENAML, 2016-09-28:
I'm still getting the same issue after changing it:

```
Traceback (most recent call last):
  File "read.py", line 12, in <module>
    reader.set_read_plan([1], "GEN2", read_power=1900)
RuntimeError: More keyword list entries (3) than format specifiers (2)
```

gotthardp, 2016-09-28:
My bad, sorry for that! (I don't have the reader today so I was unnable to test it.) Try now again, please.

gotthardp, 2016-09-29:
It should work now. Please do `git pull` and try it.

gotthardp, 2016-10-04:
Hi again. Any success with this issue?

Please note that I did some more updates and `reader.read()` now returns a list of objects rather than the list of tuples.

gotthardp, 2016-10-15:
I assume everything is OK.

closed 2016-10-15 by gotthardp