gotthardp/python-mercuryapi #4: RSSI and Antenna ID?

BrandonPotter, 2016-10-03:
Is it possible to get the RSSI and Antenna ID from the tag reads?

gotthardp, 2016-10-03:
Currently not, but I will make it possible. In few days.

BrandonPotter, 2016-10-03:
Awesome. It looked like in BuildTagData there might just be some Mercury fields that need to go into the python tuple. Don't know if it's that easy or not...

gotthardp, 2016-10-03:
It is quite easy, but I consider changing the tuple to an object, so you'd do `Tag.rssi` or `Tag.epc` to get the data. If a tuple gets too large it becomes messy.

gotthardp, 2016-10-04:
Hi again. It shall work, so please check and let me know.

BrandonPotter, 2016-10-04:
Awesome. Is it added to the tuple or changed to an obj?

gotthardp, 2016-10-04:
It's not a tuple anymore. It's an object with (currently) 4 attributes, but it's very easy to add more. This way it is more flexible and future-proof. I hope I reflected everything in the updated README.md.

BrandonPotter, 2016-10-05:
This works great! Thank you!

closed 2016-10-05 by BrandonPotter