gotthardp/python-mercuryapi #5: python2

karimi68, 2016-10-06:
Is that possible to use python2 for this python wrapper?
I am using this code in ROS Indigo, but rospy supports only python2.

gotthardp, 2016-10-07:
It should be possible. I updated the Makefile to find the right Python version automatically. Please try it and let me know if it works.

karimi68, 2016-10-07:
It doesn't work.
[image]

Based on the error I eliminated the following part of test.py (asynchronous reading mode) that I end up with the following error:

[image]

gotthardp, 2016-10-07:
Oops, sorry. I should have tested it better. Unfortunatelly, I don't have the reader here with me.

I did two more fixes, could you please check again and let me know if there is any other error?

karimi68, 2016-10-07:
I still got the same error.

TypeError: expecting a list of 8-bit integers

gotthardp, 2016-10-07:
That shouldn't have happened. Sorry for asking this, but did you pull the latest version and run `make` and `make install`?

karimi68, 2016-10-07:
Sure. Everything works properly with python3 but not 2. It's kind of weird.

gotthardp, 2016-10-07:
Weird. I will do more tests on Monday with real hardware, but `set_read_plan` parameter parsing with Python 2 works fine here.

karimi68, 2016-10-07:
OK. Thank you for your time. I appreciate it if you let me know the result on real hardware.

gotthardp, 2016-10-10:
It works for me. I installed both Python 2.7 and Python 3.5, tried `test.py` on both 32-bit and 64-bit machine and it worked fine. I really suspect your error may be caused by some dependency error, which caused you don't execute the latest code.

karimi68, 2016-10-10:
I did it over using another computer. First I got an error "fatal error: Python.h: No such file or directory" when I used "make". Then I tried the below commands to fix it:

sudo apt-get install python-dev  # for python2.x installs
sudo apt-get install python3-dev  # for python3.x installs 

Invoking these commands, the package is installed properly, and it works for python3. But, again there is an error when I try python2:
[image]

I am pretty sure that there is something wrong with my python2 dependencies.

gotthardp, 2016-10-10:
The `python-dev` should be listed as a dependency. Good point.

The package was not correctly installed for Python 2. The `python setup.py install` needs to be invoked for each Python version installed. If you use `make install` you need to set the version by `make PYTHON=python install` or `make PYTHON=python3 install`. If both versions are installed, the Python3 always takes precedence and plain `make install` doesn't install Python 2.

karimi68, 2016-10-10:
"make PYTHON=python install" solved the issue. Thank you a lot for your time.

gotthardp, 2016-10-10:
Great! Thank you for your patience. I improved the documentation to avoid similar problems in the future.

If you need anything else, just open a new issue.

closed 2016-10-10 by gotthardp