gotthardp/python-mercuryapi #8: Support getting the user data from tags

MaorGariv, 2017-03-03:
Hi,

Great job for wrapping it in Python:)

Do you have any plans to add the user defined data to [TagReadData](https://github.com/gotthardp/python-mercuryapi/blob/360a7fa3d05e3c81a0f94d05165c7caffbe244ad/mercury.c#L653) (i.e., the data that can be written to tags)?

Siceth, 2018-05-11:
@gotthardp Do you know what the major roadblock is for user memory? Like is it forseeably likely not ever possible or is it a limitation of Mercury?

gotthardp, 2018-05-12:
@Siceth I think it only needs to be implemented. There is no roadblock, except me being busy and nobody else willing to contribute.

gotthardp, 2018-09-11:
Hello. Sorry for the delay. I just implemented the read access to memory banks. See the `bank` argument of the set_read_plan.

closed 2018-09-11 by gotthardp