gotthardp/python-mercuryapi #9: Run in Raspberry Pi 3 B running Windows 10 IoT

MarianoGomezBidondo, 2017-04-21:
Hi!
I already have runing in a Python proyect in Windows 10 and it's works perfect!
But... i need to have the same running in a Windows 10 **IOT**, but when i deploy form Visual Studio 2015 (the same proyecto what it's ok over Windows 10) i have a Error "mercury module not found"... i guest what i need a Raspberry compiled version of the wrapper.... any help?

Regards!!
PD: Grate job done with the library...
PD2: Sorry my terrible english

gotthardp, 2017-04-21:
Hello. Unfortunately I have no experiences with Windows 10 IoT. I don't know how it differs from std windows.

gotthardp, 2018-02-10:
Hello. I just added (and tested) build instructions (see README.md). Please try to follow them and raise an issue if that does not work.

closed 2018-02-10 by gotthardp