gotthardp/python-mercuryapi #11: windows installer for Python 3.4 ThingMagic Mecury API

juliotesla, 2017-05-20:


luizhbernardes, 2017-08-19:
I've been trying to install on Python 3.4 through the setup-win.py but I always get this error:
error: Unable to find vcvarsall.bat
I already checked the vcvarsall.bat file and it's in this directory "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC"

So, I tryed on Python 3.5 too and the problem is different, he cant find the tm_reader.h.
Should I put the C files of the mercuryapi in some specific directory?
Because if I could get the installation files of the 3.5 I could transfer then to 3.4.

If you have some Windows installer for it, will help a lot.
Windows 10 64bit
Python3.4

Thanks!

gotthardp, 2017-08-19:
But an installer for 3.5 is provided. That doesn't work?

luizhbernardes, 2017-08-19:
Nope, it gives me an error, "Python 3.5-32 is required"

luizhbernardes, 2017-10-10:
Hi, I solved the Python 3.4 problem with the file "vcvarsall.bat" making a quick fix in this file "Python34/Lib/distutils/msvc9compile.py".

Now I'm with the following problem, even in Python 3.5:
mercuryapi-1.29.4.34/c/src/arch/win32/include\pthread.h(218): fatal error C1083: Unable to open the file include: 'time.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\bin\\cl.exe' failed with exit status 2

I'm already trying to fix the VS installation to check if I missed any important SDK component.

Did you know anything about this kind of missing file?

Thanks!

gotthardp, 2018-02-10:
Hello. I just made the installer for python 3.6 x86-64. Please check the release v0.3

closed 2018-02-10 by gotthardp