gotthardp/python-mercuryapi #12: Read Timeout

ryutatsu, 2017-06-26:
I get this error when running my code.  Things work fine until the call to read.   I'm assuming this has to do with the Tag buffer filling up for some reason, but I don't know how to clear it properly.  I've tried setting different baud rates but no luck.

```bash
M6e Nano
[u'NA2', u'NA3', u'IN', u'JP', u'PRC', u'EU3', u'KR2', u'AU', u'NZ', u'open']
Traceback (most recent call last):
  File "test.py", line 12, in <module>
    print(reader.read(timeout=1000))
RuntimeError: Timeout
```

here is the code that I am running
```python
from __future__ import print_function
import time
import mercury
reader = mercury.Reader("tmr:///dev/ttyUSB0", baudrate=115200)

print(reader.get_model())
print(reader.get_supported_regions())

reader.set_region("EU3")
reader.set_read_plan([1], "GEN2", read_power=1900)
print(reader.read(timeout=1000))

reader.start_reading(lambda tag: print(tag.epc, tag.antenna, tag.read_count, tag.rssi))
time.sleep(1)
reader.stop_reading()
```

gotthardp, 2017-06-26:
Hello. What is the reason for assuming the problem is related to buffer filling? (I'm not saying you're wrong, just asking.)

ryutatsu, 2017-06-26:
Not an educated guess as I really don't know.
It was from reading mercury.c 
/* In case of TAG ID Buffer Full, extract the tags present in buffer. */

gotthardp, 2017-06-26:
Basic question: the C-based mercury tools are reading the Tag fine, or have problems too?

ryutatsu, 2017-06-26:
Right now I've only tried using this wrapper and haven't setup up anything to use the C tools

ryutatsu, 2017-06-26:
I just tested the read.c example provided in the `mercuryapi-1.29.2.10/c/src/samples` directory.
It seems to work fine and I am able to read the tag number.

gotthardp, 2017-06-26:
To be honest, I have no idea what could be wrong.

ryutatsu, 2017-06-26:
well, with a fresh install now the python is working for me.  I'm not really sure what happened with the first round.

gotthardp, 2017-06-26:
Cool. :)

closed 2017-06-26 by gotthardp