gotthardp/python-mercuryapi #14: How to convert the list with byte strings into something useable? What is the reader.read() output?

markijzerman, 2017-07-17:
Hi, you're currently saving me a lot of work on my current project (thank you!), but I've got some issues with converting the list of byte strings. 

I want to compare a list of known ID's with the tags that are being seen, and get the location of the seen tags in the known tags-list.

I keep running into the issue that the list with byte strings doesn't seem to be a real list, and it keeps referring to mercury.TagReadData (TypeError: 'mercury.TagReadData' object is not subscriptable). Also, doing it like this (below) also does not work, the output is always false. So I think the output from reader.read() might not be a normal list but something else?

```
# importing libraries
import threading
import time 
import mercury
import collections

# set-up for the connected reader
reader = mercury.Reader("tmr:///dev/ttyS0", baudrate=115200)
print(reader.get_model()) # print the model to see if it's working
reader.set_region("EU3") # set a region to work with
reader.set_read_plan([1], "GEN2", read_power=1900)

# Below should be the IDs of the RFID tags!
knownIDs = [b'E2000015860E01451120AB56',
        b'48656C6C6F2101591120AB75',
        b'E2000015860E01601120AB6E',
        b'E2000015860E01611120AB76']

def readTags():
	threading.Timer(0.15, readTags).start()
	# this reads all of them in one go within the given time limit, and puts them in the CurIDs list
	incomingIDs = reader.read(timeout=100)
	print (list(incomingIDs))
	print ("length of list is", len(incomingIDs))

	# comparing lists, output the location of the found IDs within knownIDs
	print (b'E2000015860E01451120AB56' in incomingIDs)

readTags()
```

Thanks in advance!

markijzerman, 2017-07-17:
Ah, I now get that they're objects instead of flat items. Still working on how to convert this though...

markijzerman, 2017-07-17:
Fixed it like this. Not so elegant but will think of a better way later.

```
# importing libraries
import threading
import time 
import mercury
import collections

# set-up for the connected reader
reader = mercury.Reader("tmr:///dev/ttyS0", baudrate=115200)
print(reader.get_model()) # print the model to see if it's working
reader.set_region("EU3") # set a region to work with
reader.set_read_plan([1], "GEN2", read_power=1900)

# Below should be the IDs of the RFID tags! Their location is the page numbers
knownIDs = [b'E2000015860E01451120AB56',
        b'48656C6C6F2101591120AB75',
        b'E2000015860E01601120AB6E',
        b'E2000015860E01611120AB76']

def readTags():
	threading.Timer(0.15, readTags).start()
	# this reads all of them in one go within the given time limit, and puts them in the CurIDs list
	incomingIDs = reader.read(timeout=100)
	#print (list(incomingIDs))
	print ("length of list is", len(incomingIDs))
	# make list out of all the tags epc item. CurIDs is flat list
	curIDs = []
	pageNumbers = []
	for x in range(0, len(incomingIDs)):
		curIDs.append(incomingIDs[x].epc)

	# comparing lists, output the location of the found IDs within knownIDs
	for x in range(0, len(curIDs)):
		pagePresent = knownIDs.index(curIDs[x])
		print (pagePresent)

readTags()

```

closed 2017-07-17 by markijzerman

Knells1, 2019-08-27:
have you found any other way yet? i can use your way but it seems to give an error when the read tag is not in the knownIDs list

markijzerman, 2019-08-27:
Hi! I ended up just doing it like this. Haven't done a lot of Python lately, and I don't remember what the error is, but you could try to catch it with an Except: https://www.w3schools.com/python/python_try_except.asp

Knells1, 2019-08-27:
awesome thanks, let me get on it and see.

Knells1, 2019-08-27:
> Hi! I ended up just doing it like this. Haven't done a lot of Python lately, and I don't remember what the error is, but you could try to catch it with an Except: https://www.w3schools.com/python/python_try_except.asp

yes it might not be the best solution but it helps the situation. thanks

gotthardp, 2019-08-27:
A more elegant solution could be
```python
epcs = set(map(lambda t: t.epc, incomingIDs))
print (b'E2000015860E01451120AB56' in epcs)
```
That should work.

Knells1, 2019-08-27:
yes @gotthardp it give you true or false output values