gotthardp/python-mercuryapi #15: Setup Problems

ljc19800331, 2017-10-10:
I downloaded the API from ThingMagic, and have it in the same folder as the github package you posted. I'm in Python 3.63 via Pycharm and when I try to run the setup folders： usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: no commands supplied

then when I try to run the test.py:  import mercury
ImportError: No module named 'mercury'

I'm on a 64 bit OS with Windows 10 and I'm new to coding.. 

Thanks for the help.

gotthardp, 2017-10-10:
The build process is VERY tricky.

ysaipavan99, 2018-02-10:
Could you please elaborate on the build process?

gotthardp, 2018-02-10:
Hello. I just added (and tested) build instructions (see README.md). Please try to follow them and raise an issue if that does not work.

closed 2018-02-10 by gotthardp