gotthardp/python-mercuryapi #16: Windows installer for Python 3.6

luizhbernardes, 2017-10-27:
Hi, I tried the 3.5 installer and it works, but we need a 3.6 installer for our project.
Do you have this 3.6 installer or do you know how to make it works without create a parallel 3.5 service?

Thanks!

ysaipavan99, 2018-02-10:
Hey! I'm trying with 3.5 installer and it got installed but it still shows error when i run "setup.py install" 
it says"running build
running build_ext
building 'mercury' extension
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ibuild/mercuryapi/include -IC:\Users\HP\AppData\Local\Programs\Python\Python36\include -IC:\Users\HP\AppData\Local\Programs\Python\Python36\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\8.1\include\shared" "-IC:\Program Files (x86)\Windows Kits\8.1\include\um" "-IC:\Program Files (x86)\Windows Kits\8.1\include\winrt" /Tcmercury.c /Fobuild\temp.win-amd64-3.6\Release\mercury.obj
mercury.c
mercury.c(23): fatal error C1083: Cannot open include file: 'tm_reader.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.exe' failed with exit status 2"
could you please help me with that?

gotthardp, 2018-02-10:
Hello. I just made the installer for python 3.6. Please check the release v0.3

closed 2018-02-10 by gotthardp