gotthardp/python-mercuryapi #17: Build Process in Windows 10

PandaQQ, 2018-02-02:
Hi, I'm really interesting with how to build mercury api c library in windows.  But the windows installer is not open source. I'm doing the research on how to write mercury api in C/C++. I'm not sure whether you can share the step how to build the library for windows? Thanks.

gotthardp, 2018-02-02:
Hi Panda, even the installer is open-source, but it's not so easy to setup all paths in Visual Studio. All you need to do is to run `python setup-win.py bdist --format=wininst` 
https://github.com/gotthardp/python-mercuryapi/blob/master/setup-win.py. The trick is really just the MSVS config, which is machine specific so I cannot help with that, unfortunately.

PandaQQ, 2018-02-03:
Thanks~Already help me a lot~

closed 2018-02-03 by PandaQQ

PandaQQ, 2018-02-14:
The reason that I want to write mercury api in C/C++ just want to build node.js module. But I find a another way is (edge.js + C# + electron). And proven it can work that connect and read tags from ThingMagic RFID Rear M6e. May It can help someone who want to develop on electron desktop for windows. And I will share my code in future~