gotthardp/python-mercuryapi #23: Read_count staying at 1

haroutisonfire, 2018-08-10:
Hello, When I run continous read mode the read_count for the tag remains at one, any chance this can be fixed? I'm using the test.py code.

gotthardp, 2018-08-11:
Are you sure this is a bug in the python library? Have you tried doing the same with the mercuryapi binaries?
The read_count is directly provided by mercuryapi. If you are getting 1, perhaps the interrogation period is too short, the tag is too far or faulty. Or it could be a problem of the reader.

DanWehr, 2019-01-15:
I suspect this isn't actually an issue given the use of continuous mode. I think the Mercury API treats the read_count value as "how many reads of this tag does this object represent," not "how many times have I read this tag since continuous reading began," which @haroutisonfire may be wanting.

If it's not continous, multiple tag reads may be combined before being returned to your code and so you may see a read count higher than 1. You would expect to see this with a long read timeout and tags that respond often. If you're doing continuous reads each read may be reported to you immediately with your code being able to receive and process it before the next read comes in, so it never has the chance to combine reads of the same tag.

gotthardp, 2019-06-13:
Thanks, @DanWehr. I am closing this now as it doesn't sound like an issue.

closed 2019-06-13 by gotthardp