gotthardp/python-mercuryapi #25: ImportError: DLL load failed

riebschlager, 2018-08-17:
I'm using Windows 10 and Python 3.6.6.

Installation from `mercuryapi-0.3.win32-py3.6.exe` completes with no errors, but when attempting to import the `mercury` package, I get the following error:

`ImportError: DLL load failed: The specified module could not be found.`

What (probably stupid) thing am I doing wrong here?

gotthardp, 2018-08-18:
This is not easy to debug, unfortunately. Would you be able to use the http://www.dependencywalker.com, open the "PYD" file that got installed in the LIB directory of your Python and see if/what DLL dependency is missing?

riebschlager, 2018-08-19:
Here's a CSV output from Dependency Walker. It seems to be missing quite a bit. Are all the build dependencies in the installation section of your readme required to run the lib as well?
[mercury.csv.txt](https://github.com/gotthardp/python-mercuryapi/files/2300410/mercury.csv.txt)

gotthardp, 2018-08-20:
Hello. The API-MS are false positives, but it seems you are missing the Microsoft C++ libraries. Please install https://www.microsoft.com/en-us/download/details.aspx?id=5555
Then check if it's any better and if not, please rerun the Dependency Walker.

riebschlager, 2018-08-20:
That did it! Thanks for your help.

This is probably unrelated, but after creating a `Reader` object, I seem to be missing a lot of methods.

Here's the output of `dir(reader)`:

`['__class__', '__delattr__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'get_model', 'get_supported_regions', 'read', 'set_read_plan', 'set_region', 'start_reading', 'stop_reading']`

And the output of `reader.get_model()` is `'M6e Micro USBPro'`

gotthardp, 2018-08-20:
Cool. I updated the documentation. Thanks for your help.

Yes, a lot of methods are missing. The python wrapper does not implement all functions that are available in the Mercury API. Feel free to open a new issue if you need a specific method.

closed 2018-08-20 by gotthardp