gotthardp/python-mercuryapi #30: 'out_ltkc.h' is missing when compiling

stedio, 2018-09-25:
Hi,

I used the mercuryapi-1.31.0.33 and visual studio 2017 to compile the code. It came with the error that out_ltkc.h is missing, may I ask how to solve this? thanks.

> C:/Apps/mercuryapi-1.31.0.33/c/src/api/ltkc_win32/inc\ltkc.h(32): fatal error C1
> 083: Cannot open include file: 'out_ltkc.h': No such file or directory

stedio, 2018-09-25:
Solved by back to 'mercuryapi-1.29.4.34‘.
For no reason, the latest version removes this file.

gotthardp, 2018-09-25:
Hello. One step was missing from the build instructions:
* Go to mercuryapi-1.31.0.33\c\src\api\ltkc_win32 and run `gencode.bat`

The new instructions work with the mercuryapi-1.31.0.33.

closed 2018-09-25 by gotthardp