gotthardp/python-mercuryapi #31: ImportError : No module name "setuptools'

haroutisonfire, 2018-10-09:
Hello, I'm getting an error when calling the make command. The make process ends up with the following error:

Traceback (most recent call last) : 
    File "setup.py", line 3, in <module>
      from setuptools import setup Extension
ImportError: No module named 'setuptools'
Makefile:7: recipe for target 'all' failed
make: *** [all] Error 1

gotthardp, 2018-10-09:
You need to install the "setuptools", e.g. via `sudo apt-get install python-setuptools`

haroutisonfire, 2018-10-09:
Im still getting the same error

gotthardp, 2018-10-09:
Are you using Python 2 or 3. For python 3 you need `sudo apt-get install python3-setuptools`.

haroutisonfire, 2018-10-09:
Okay now I get fatal error: Python.h: No such file or directory 
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1 
makefile10: recipe for target 'install' failed
make: *** [install] Error 1

closed 2018-10-09 by gotthardp