gotthardp/python-mercuryapi #32: ImportError: No module named 'mercury'

haroutisonfire, 2018-10-09:
Im getting 'ImportError: No module named 'mercury' when running python test.py

gotthardp, 2018-10-09:
The module isn't installed properly. Did you run `sudo python3 setup.py install` or `sudo make install`?

haroutisonfire, 2018-10-11:
Fixed : It works when I run python3 test.py

closed 2018-10-11 by gotthardp

matiasz8, 2018-11-22:
Maybe you should type: "sudo make install", and then execute:

python test.py or
python3 test.py   

Try boths!

garysenter, 2019-08-14:
I am stuck at ImportError: No module named 'mercury'.

I did the instructions for installation, then the manual instructions. and get hung up at the CD python-mercuryapi step as that directory does not exist. What am i doing wrong?

gotthardp, 2019-08-14:
The `python-mercuryapi` directory is what you get after `https://github.com/gotthardp/python-mercuryapi.git`. If the problem persists, please open a new issue and let me know what OS you have and what exactly you did.