gotthardp/python-mercuryapi #36: Module not found error on a Raspberry Pi (This could be user error)

kodalisitaram, 2018-11-04:
Hi,
    I have followed the installation instructions on Linux and everything ran successfully without any errors. Here are list of instructions I executed.
1. sudo apt-get install patch xsltproc gcc libreadline-dev python-dev python-setuptools
2. git clone https://github.com/gotthardp/python-mercuryapi.git
3. cd in to dir,
4. make

`pi@raspberrypi: make -C mercuryapi-1.31.0.33/c/src/api
make[1]: Entering directory '/home/pi/TM/PY/python-mercuryapi/mercuryapi-1.31.0.33/c/src/api'
SOURCE_DIR=lib PATCH_DIR=lib XML_DIR=lib sh lib/install_LTKC.sh NULL lib
lib/LTK directory already exists
cd lib/LTK/LTKC/Library; make CC="cc -Ilib/LTK/LTKC/Library -Ilib/LTK/LTKC/Library/LLRP.org -I. -g -Wall -fPIC" STRIP="strip"
make[2]: Entering directory '/home/pi/TM/PY/python-mercuryapi/mercuryapi-1.31.0.33/c/src/api/lib/LTK/LTKC/Library'
cd LLRP.org; make all
make[3]: Entering directory '/home/pi/TM/PY/python-mercuryapi/mercuryapi-1.31.0.33/c/src/api/lib/LTK/LTKC/Library/LLRP.org'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/pi/TM/PY/python-mercuryapi/mercuryapi-1.31.0.33/c/src/api/lib/LTK/LTKC/Library/LLRP.org'
make[2]: Leaving directory '/home/pi/TM/PY/python-mercuryapi/mercuryapi-1.31.0.33/c/src/api/lib/LTK/LTKC/Library'
make[1]: Leaving directory '/home/pi/TM/PY/python-mercuryapi/mercuryapi-1.31.0.33/c/src/api'
mkdir -p build/mercuryapi/include
find mercuryapi-*/c/src/api -type f -name '*.h' ! -name '*_imp.h' | grep -v 'ltkc_win32' | xargs cp -t build/mercuryapi/include
mkdir -p build/mercuryapi/lib
find mercuryapi-*/c/src/api -type f -name '*.a' -or -name '*.so.1' | xargs cp -t build/mercuryapi/lib
/usr/bin/python3 setup.py build
running build
running build_ext`

Once its done, When  I run the /usr/bin/python3 test.py, I get the import error: No module named mercury error. I am trying to connect a ThingMagic rfid reader to a Raspberry Pi and read some tags.

gotthardp, 2018-11-07:
Did you run `make install`?

kodalisitaram, 2018-11-29:
Hi, sorry for late response. Yes I did. I will try it again.

DanWehr, 2018-11-29:
I just tried this project today for the first time on the Raspberry Pi, and may be able to shed some light on what @kodalisitaram ran into.

On my Pi I have both Python 2 and 3 installed, and the program I want to use the Mercury API in is run through Python 2. When I ran `sudo make` and `sudo make install` it installed the mercury package for Python 3 specifically, and of course when I tried to run my code through Python 2 the module was missing. 

As stated at the end of the repo readme, you can instead use `sudo make PYTHON=python` to build for Python 2, but there's another important step! If you simply run `sudo make install` again as the doc says, it will still do the install through Python 3.

The readme says the make install is a shortcut to the setup.py file, which you will see has a python3 shebang. The solution is then to skip `make install`, and directly run `sudo python setup.py install`, which will put the package in the right place. Doing that, I have this project now working and talking to a networked reader is successfully producing tag reads.

gotthardp, 2018-11-30:
Thank you, @DanWehr. I will improve the documentation based on your findings.

romainrobotics, 2019-03-07:
When using this library, has anyone gotten it to work with code completion in an IDE such as PyCharm?

kodalisitaram, 2019-04-14:
@DanWehr and @gotthardp For your inputs. Sorry it took me a while to respond. I will try it now and close this issue right away. Thanks a bunch.

BoomerIsSpartacus, 2019-06-11:
Helo. Is it possible to read the phase and frequency parameters with this api for python? If not, how can this feature be added?

BoomerIsSpartacus, 2019-06-11:
So it is possible to read phase, but it seems that changing the hopTime and hopTable is not possible. Is this correct?

gotthardp, 2019-06-11:
@BoomerIsSpartacus, please don't reuse old issues for reporting new ones. Instead, open a new ticket, one for each issue you have.

gotthardp, 2019-06-13:
The documentation has been updated and the hop time/table has been addressed in #64.

closed 2019-06-13 by gotthardp

BoomerIsSpartacus, 2019-06-13:
Thank you very much. I appreciate it.