gotthardp/python-mercuryapi #37: reader.Read() invalid argument if bank is set

florianhumblot, 2018-11-21:
I've gotten to the point where I'm able to read the tags using the reader.Read method.

I create a reader using `mercury.Reader("tmr:///COM6")`
I set the region to "EU3"
I set the readplan: 
`rfid_reader.set_read_plan(rfid_reader.get_antennas(), "GEN2")`
This works fine, I'm able to get some information from the API for example:

`b'303A5E59000000000000DB96'` 

However, I'd like to get more of the information, so I tried adding the bank parameter to the `set_read_plan` function like so:
`rfid_reader.set_read_plan(rfid_reader.get_antennas(), "GEN2", "tid")`

I've also tried adding a list of 1 and a list of multiple bank parameters like so:
`rfid_reader.set_read_plan(rfid_reader.get_antennas(), "GEN2", ["tid"])`
and
`rfid_reader.set_read_plan(rfid_reader.get_antennas(), "GEN2", ["tid", "reserved"])`

However, as soon as I set a bank parameter, I get the following error:

```
Traceback (most recent call last):
  File "C:\ti-software\RFID\rfid\app.py", line 38, in start_action
    self.read_timer.timeout.connect(self.update_list())
  File "C:\ti-software\RFID\rfid\app.py", line 45, in update_list
    tags = utils.rfid_read(self)
  File "C:\ti-software\RFID\rfid\utils.py", line 42, in rfid_read
    temp = mainWindow.rfid_reader.read()
RuntimeError: Invalid argument
```

just in case it matters, the reader is kept as a property of mainWindow.

I'm using the latest 32 bit release for python 3.6.

Any idea what is going on? Or what I should change to get more data back ?

thanks!

florianhumblot, 2018-11-22:
Thanks for the quick response, 

I've tried editing the code to become: 
`rfid_reader.set_read_plan(mainWindow.rfid_reader.get_antennas(), "GEN2", bank=["user"])` 
and 
`rfid_reader.set_read_plan(mainWindow.rfid_reader.get_antennas(), "GEN2", bank=["user", "epc", "reserved", "tid"])`

but still no luck, I still get the `RuntimeError: Invalid argument` for the `reader.Read()` function call.

Do you have another suggestion?

gotthardp, 2018-11-22:
Did you try also my example? Without the get_antennas() construct?
```python
reader.set_read_plan([1], "GEN2", bank=["user"])
```

florianhumblot, 2018-11-22:
Yes, I've tried that too, 
I still get the exact same error of `RuntimeError: Invalid argument` in the `reader.Read()` function call

Is there a way to more deeply debug this?

gotthardp, 2018-11-22:
are you working on linux on windows? adding debug in linux is quite easy

florianhumblot, 2018-11-22:
> 
> 
> are you working on linux on windows? adding debug in linux is quite easy

I'm currently developing an application on windows.

florianhumblot, 2018-11-29:
Updating the issue:

Running on linux (ubuntu 18.04) I don't get the `RuntimeError: Invalid argument` error anymore. So that seems to be a platform specific issue, I am running windows 10 professional with the latest updates in case you were wondering about that.

However, even when selecting every single bank, I only get the byte string with the EPC number, i.e.:
`[b'303A5E5900000000000000000DB9D']`
That is what I get when printing the output of:

`rfid_reader.read(timeout=450)` 

Is there anything else I'm supposed to be doing to get the databanks ?

Thanks for your help and patience!

florianhumblot, 2018-11-29:
Just tested the bank argument on linux (ubuntu 18.04) and I get the same error:

`rfid_reader.read(timeout=450)`
`RuntimeError: Invalid argument`
`Segmentation fault`

Followed by the program crashing. 

You said debugging on linux would be easy, I'd love some pointers as to how to add that debugging to figure out what's going on.

florianhumblot, 2018-11-29:
Running the program through GDB I get the following output:

```
Thread 5 "Thread (pooled)" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd6cc4700 (LWP 3586)]
0x00007fffee2bc6a2 in TMR_SR_addTagOp (reader=0xf398c0, tagop=0x7fffffffc690, rp=0xf398d8, msg=0x7fffd6cc30e0 "(\027\"", j=0x7fffd6cc2ca4 "\002", readTimeMs=450, byte=0x7fffd6cc2d20 "\177") at serial_reader.c:5817
5817        TMR_SR_msgAddGEN2WriteTagEPC(msg &i, 0, args->epcptr->epc, args->epcptr->epcByteCount);
```

Does this mean anything to you ?

gotthardp, 2018-11-29:
I cannot reproduce this. Which MercuryAPI version are you using? I have "mercuryapi-1.31.0.33".

florianhumblot, 2018-11-29:
Hey, 

I'm using the automatically downloaded version (the one that the linux installation downloads) which is `mercuryapi-1.31.0.33' 

Thanks for taking the time to debug this.

gotthardp, 2018-11-29:
I made a fix that could help? Would you be able to pull the latest master, build and install the latest code. If it works, I will compile it on windows and do a release.

florianhumblot, 2018-11-29:
Hey, thanks for the fix! 
I don't get the error or crash anymore. 

However, when I print the output of `reader.read()` like so:

```
print(rfid_reader.read(timeout=450))
```

It returns:
```
[b'303A5E590000000000000000DB89', b'303A5E5000000000000000000DC1', ...]
```
where the ... is another few tags.
Shouldn't it be returning a list per tag containing the data? 

The readplan is now setup like so:

```
rfid_reader.set_read_plan(rfid_reader.get_antennas(), "GEN2", bank=["user", "epc, "tid", "reserved"])
```

florianhumblot, 2018-11-29:
Awesome, thank you very much for the time and effort you've put into this! 

I'll go ahead and close this issue as it's been solved.

closed 2018-11-29 by florianhumblot

gotthardp, 2018-11-29:
I am glad it works.