gotthardp/python-mercuryapi #39: reader.start_reading

haroutisonfire, 2018-11-29:
Im having issue running the reader.start_reading call function when setting the read plan to include for the bank arguments. For example when I set: reader.set_readplan([1], "GEN2", bank=["epc", "reserved"], read_power=1400), the reader.start_reading call returns nothing. When I get rid of the bank argument it starts to work however the bank arguments will print as "None" such as the reserved, tid etc.

gotthardp, 2018-11-29:
Yeah. The bank retrieval is not supported for the continuous reading. This is how the MercuryAPI works. They assume the user first reads all tags (e.g. by the continuous reading) and then makes a targeted request to read the bank data. I think this is because the operation is quite time&energy consuming.

gotthardp, 2018-11-29:
The read() is synchronous and can read the bank data. The start_read() is asynchronous, so your python program can continue and do other tasks while scanning. If you don't need asynchronicity, then the start_read() is of no use for you.

rasidmusic, 2019-02-19:
@gotthardp Can you post a sample code on how to read user data with the bank parameter?
My Code returns EPC, despite setting the bank in the set_read_plan:
```
reader.set_read_plan([1], "GEN2", bank=["user"], read_power=1400)
allTags.extend(reader.read(timeout=450))
```
#49

gotthardp, 2019-02-19:
The following works for me
```python
#!/usr/bin/env python3
from __future__ import print_function
import mercury
reader = mercury.Reader("tmr://dev/ttyUSB0", baudrate=115200)

reader.set_region("EU3")
reader.set_read_plan(reader.get_antennas(), "GEN2", "user")
for tag in reader.read(timeout=450):
    print("EPC", tag.epc)
    print("Antenna", tag.antenna)
    print("Count", tag.read_count)
    print("RSSI", tag.rssi)
    print("user", tag.user_mem_data)
```

rasidmusic, 2019-02-19:
> The following works for me
>
> ```python
> #!/usr/bin/env python3
> from __future__ import print_function
> import mercury
> reader = mercury.Reader("tmr://dev/ttyUSB0", baudrate=115200) [...]

Thank you. This works. However my Code does not return a correct result. Can you check #49 please?

rasidmusic, 2019-02-20:
Hey @gotthardp 
I tried your code. Unfortunately I get the following results:
```
EPC b'E200001963020091231027DE'
Antenna 1
Count 7
RSSI -37
user bytearray(b'L18100117\r\n\x00\x00\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
EPC b'E2000017230C01961100AA8E'
Antenna 1
Count 7
RSSI -31
user None
```
As you can see, the second tag does not return user data. That is, if two tags are read at the same time. Both show the correct user data if read seperately.

jhuncamit, 2019-04-12:
Hi can you post a sample code for the start_reading and what response are you getting? Thanks!

gotthardp, 2019-04-12:
The https://github.com/gotthardp/python-mercuryapi/blob/master/test.py contains a sample start_reading code. Do you need something more?

jhuncamit, 2019-04-12:
Yeah sorry, let me rephrase my question. Basically what I need is a fast continuous read from the m6e that returns tags and its rssi. Can you help me with that? Thanks!

jhuncamit, 2019-04-12:
When I ran the test.py, I'm getting tag reads however it seems to be inconsistent and doesn't read all of the tags I placed on top of the antenna.

gotthardp, 2019-04-12:
You may need more antennas or multiple read attempts if there are many tags. But this is beyond this library.

gotthardp, 2019-06-11:
There was a bug fixed in the code, so the original issue has been addressed.
Please open a new issue (one ticket per issue) if you continue having troubles.

closed 2019-06-11 by gotthardp