gotthardp/python-mercuryapi #41: Error when using with USB

dannielperez, 2018-12-13:
I tried a test using the ThingMagic astra through IP and it worked, switched to USB and I get this error when running.
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    reader = mercury.Reader("tmr:///dev/ttyUSB0", baudrate=115200)
TypeError: Timeout

gotthardp, 2018-12-13:
This appears to be a connectivity issue rather than a software bug.

callen5914, 2019-02-16:
That's it?

Seems pretty vague. I too ran this code and it gives me the timeout error as well. 

Running Ubuntu(latest)

Hooked up to the Sparkfun Simultaneous Reader Writer through usb/FTDI cable.

gotthardp, 2019-02-18:
I really cannot help here. The Python only displays the error given by Mercury API. The error "Timeout" is really very vague, so I can only say it is a connectivity problem rather than a bug.
If you want to try debugging this issue, download the latest API for C (https://www.jadaktech.com/documentation/rfid/mercuryapi/#mercury-api-v1-31-2), build it and try executing some of the examples. Do you get the same error, or native C examples work?

florianhumblot, 2019-03-01:
Make sure you've got the right USB port. 
On my ThingMagic m6e I have two micro-USB ports, one of which does work with the API and the other does not and results in a TimeOut as well.

gotthardp, 2019-06-11:
Closing now. Please provide more debugging info if this issue is still valid.

closed 2019-06-11 by gotthardp