gotthardp/python-mercuryapi #42: Cannot build on Mac OS

AaronHuus, 2018-12-14:
When running `make` from on Mac OS, I get a lot of the following errors:

```
./ltkc_base.h:142:5: error: unknown type name 'llrp_u16_t'; did you mean 'llrp_u16v_t'?
```

Is building/installing from MacOS supported? And if not, what is required to allow for this?

gotthardp, 2018-12-14:
I never tried Mac OS, but it should work. The MercuryAPI contains special build instructions for IOS (see REAME.IOS in mercuryapi-1.31.0.33/c). The wrapper is a tiny layer above the mercuryAPI, so the build problems usually come from the MercuryAPI. I’d recommend you to build the API itself, then copy the .h and .a/.so somewhere and run the python build. For Windows and Linux it is automated, but Mac may need some tweaks before it works.

rhaag71, 2019-03-01:
@AaronHuus Did you ever figure this out, I am running into the same issue.

vssystemluba, 2019-04-08:
Bumping this; I'm hitting the same issue!  Anybody made any progress?

gotthardp, 2019-06-13:
@AaronHuus has made this `setup.py` that might work. Anyone willing to try this?
https://github.com/AaronHuus/python-mercuryapi/blob/feature/mac-os-support/setup.py

AaronHuus, 2019-06-20:
I do not believe I ever fully got this working before being pulled on to something else

Moebe1, 2019-07-20:
Hi I am happy to help out with the mac issue. I have one that is not doing anything at the moment. I can have it ready in a day or two if needed.

gotthardp, 2019-07-23:
Hello everyone, using the machine from @Moebe1 I was able to build the code under MacOS X and install the Python extension.
When you have some time, please check if it works for you. Just download the fresh master (or pull and remove all the downloaded files) and try `python setup.py build install`.

closed 2019-08-06 by gotthardp

ddelano34, 2020-09-25:
I know this is an older bug, but I am running into an issue building on MacOS X 10.15.6 with mercuryapi-1.31.4.35. I followed the instructions in the readme but get an 'unknown type name' error and an 'architecture not supported' error. Which version of Mac OS and Mercury API was the patch working on?

Most of the errors seem to come from the pthread.h file.
`
In file included from mercury.c:23:
In file included from build/mercuryapi/include/tm_reader.h:55:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include/pthread.h:55:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include/_types.h:27:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include/sys/_types.h:55:9: error: 
      unknown type name '__int64_t'
typedef __int64_t       __darwin_blkcnt_t;      /...
`

ddelano34, 2020-09-29:
Finally figured out what the issue was! This is an issue with the XCode 12 version of python3 setting the -arch arm64 flag during the compile of the extension. I used homebrew to install official version of Python and it works.
This github issue led me to the solution: https://github.com/ycm-core/YouCompleteMe/issues/3770#issuecomment-700436195

Hopefully this will save someone else some time trying to figure out this issue.