gotthardp/python-mercuryapi #43: reader.stop() Timeout issue

rasidmusic, 2018-12-19:
Hey,

I am running a flask based web application. In this apllication when visiting a specific page, the readprocess starts as a background thread.
If the index page is visited, the read should be stopped.
If the reader is busy reading, I can't stop the reader with reader.stop(). I get the following error massage:
`Traceback (most recent call last):
  File "manStop.py", line 7, in <module>
    reader = mercury.Reader("tmr:///dev/ttyUSB0", baudrate=115200)
TypeError: Timeout
`
Any ideas?

gotthardp, 2018-12-28:
The error appears to be thrown at (re)instantiation of the reader, not at reader.stop(). Could it be a bug in your code?

rasidmusic, 2019-01-07:
Hey, 
yeah you're right. It does not happen on reader.stop_reading(). The Timeout exception is thrown at
reader = mercury.Reader(XYZ) if a read process is ongoing.
Do you have a clever on how to stop a ongoing reading process?

gotthardp, 2019-01-07:
the read() function has a timeout. it cannot be interrupted

rasidmusic, 2019-02-19:
For everyone that has similar issues, here is what the problem ist/how to solve it:
It is not possible to define `reader = mercury.Reader("tmr:///dev/ttyUSB0")` if a read is ongoing. The channel is being used and therefore blocks the new connection.
The read has to be stopped correctly before reconnection with `reader.stop_reading()`.
However sometimes it takes some time until the read is stopped. In my case, the read function is a very complex one. So i wrapped my entire read process inside a thread and gave it an attribute. If this attribute is set to False, the read will be stopped with reader.stop_reading() and just to be sure it is really stopped, the entire Thread is stopped. But this is a custom solution I guess.

closed 2019-02-19 by rasidmusic