gotthardp/python-mercuryapi #45: Referencing multiple USB readers

clickworkorange, 2019-01-16:
First, a big thank you for providing this library - it's really awesome!  Unfortunately I've become stuck on a curious issue: I've got two Mercury M6e readers connected via USB, with a UDEV rule* assigning fixed references for each one - yet in a simple Python test the readers keep swapping around, which makes it impossible to identify which antenna read which tag! This is my test program: 

    import mercury
    from time import sleep

    readerA = mercury.Reader("tmr:///dev/MercuryA", baudrate=9600)
    readerB = mercury.Reader("tmr:///dev/MercuryB", baudrate=9600)

    def detectA(tag):
      print tag.epc + " A" + str(tag.antenna)

    def detectB(tag):
      print tag.epc + " B" + str(tag.antenna)

    readerA.set_region("EU3")
    readerA.set_read_plan([1,2,3,4], "GEN2", read_power=800)

    readerB.set_region("EU3")
    readerB.set_read_plan([1,2,3,4], "GEN2", read_power=800)

    try:
      readerA.start_reading(detectA)
      readerB.start_reading(detectB)
      while True:
            sleep(0.1)

    except(KeyboardInterrupt, SystemExit):
      readerA.stop_reading()
      readerB.stop_reading()

And here is some sample output, with a single tag resting directly on top of one of the antennas: 

    300ED89F335000400182120B A2
    300ED89F335000400182120B A2
    300ED89F335000400182120B B2
    300ED89F335000400182120B A2
    300ED89F335000400182120B A2
    300ED89F335000400182120B B2
    300ED89F335000400182120B A2
    300ED89F335000400182120B B2
    300ED89F335000400182120B A2
    300ED89F335000400182120B A2
    300ED89F335000400182120B A2
    300ED89F335000400182120B A2
    300ED89F335000400182120B A2
    300ED89F335000400182120B B2
    300ED89F335000400182120B B2
    300ED89F335000400182120B A2
    300ED89F335000400182120B A2
    300ED89F335000400182120B A2
    300ED89F335000400182120B B2

As you can see, readerA randomly switches between calling the detectA() and detectB() callbacks! How can this be, when I've separately defined each reader, and its callback function!? 

*) My UDEV rule (/etc/udev/rules.d/99-usb-serial.rules), for those playing along at home: 
    SUBSYSTEM=="tty", KERNELS=="1-1.1.2:1.0", SYMLINK+="MercuryA"
    SUBSYSTEM=="tty", KERNELS=="1-1.1.3:1.0", SYMLINK+="MercuryB"

gotthardp, 2019-01-16:
That is weird indeed. Could you please try one more thing: first start both A+B, but then after some cycles stop B and keep only A running. Does the random alternation stop, or not?

clickworkorange, 2019-01-16:
Thanks Petr! I changed the detectA() function to stop readerB after ten reads, and this stops any calls to detectB(): 

    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 B1
    300ED89F3350004001821268 B1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 B1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 B1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 B1
    Stopping readerB
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1
    300ED89F3350004001821268 A1

Note that antenna "B1" is some four metres away from antenna "A1", and antennas "A2-4" are in between, so it cannot be simply that "the wrong antenna" sometimes reads the tag.

gotthardp, 2019-01-16:
But-- if both readers can see the tag, when you start both A+B, each will CONCURRENTLY use its antenna and read the tag, i.e. the prints from detectA (called by A) and detectB (called by B) will be mixed up. Could you please explain again what is wrong with this behaviour,

clickworkorange, 2019-01-16:
> if both readers can see the tag

They cannot, and should not - each tag can only be read once in my application, and it is critical that I know where (i.e. which reader and antenna) it was first seen.

gotthardp, 2019-01-16:
So you say that the tag is so far from one of the readers that only one of the readers can "see" it (from the radio perspective). If both readers are on one table the tag can be easily seen by both readers. This is obvious, right?

clickworkorange, 2019-01-16:
Hang on a minute - the plot thickens! In the interest of eliminating all possibilities I disconnected the antenna from A4, and guess what; the A4 callbacks stopped coming! Instead I see some (much less frequent) A3 triggers. Hmmmm.

gotthardp, 2019-01-16:
I cannot find any possible root-cause in the code. Could you please print `tag.rssi` as well? Perhaps the difference of RSSI from different antennas could shed some more like onto this.

gotthardp, 2019-01-16:
I am happy it's not a bug :) Readers transmit energy and then receive the backscattered signal. I am not an RF expert, but I see no reason why reader B couldn't receive the backscattered signal triggered by the reader A. Or, if two readers supply energy there can be more of the backscattered signal, which could explain what you experienced.

clickworkorange, 2019-01-16:
Yes, it all makes sense; each reader will iterate between its four antennas (I think the default dwell time is 250ms), and any tags present are energised by one antenna at a time; the problem is not the antenna pulse itself, but that the same # antenna on the second reader is listening to the response from the tag at the same time. Had it been an eight channel reader I assume this wouldn't be a problem, but now I have to find a way to sequence the antennas between the two readers. Currently running a test of 1000 reads from all four readerB antennas, with increasing delay before starting readerA, and logging the number of errors for each delay (50ms to 1s in 10ms increments). Perhaps this simple approach will be sufficient. Will report back with my findings.