gotthardp/python-mercuryapi #46: Read data banks fails on networked reader

DanWehr, 2019-01-16:
I'm currently attempting to read tag data banks with a networked Sargas reader, and the current implementation is failing to read the data banks. For example simply running:

```python
reader = mercury.Reader("tmr://192.168.1.101")
reader.set_read_plan([1], "GEN2", bank=["epc"], read_power=3000)
tags = reader.read(timeout=1000)
for tag in tags:
    print(tag.epc, tag.antenna, tag.read_count, tag.rssi, tag.epc_mem_data)
```

All of the tag parameters are correct, except for epc_mem_data which is None for every tag returned. The same None result applies for any other data bank, or if I'm attempting to get multiple instead of one.

After a little digging the source of the problem may be because the current bank read is implemented using embedded tag ops set in the read plan, combined with me using a networked reader accessed through an IP, not a serial connected reader. This is touched on in the MercuryAPI ProgrammerGuide:

> Embedded TagOps are only supported with Gen2 (ISO18000-6C) protocol tags and when connected to readers of type SerialReader and RQLReader type M6.

This makes me think that for a networked reader I need to use direct invocation through Reader.ExecuteTagOp() that is mentioned in the programmer guide. I may take a stab at adding a method for this myself if I'm correct here.

To add an additional layer of confusion, the official Universal Reader Assistant program (which I presume still uses the Mercury API in the background) has an "Embedded ReadData" option that _does_ work with my networked reader and the tags I'm testing with, which is in conflict with the above quote and leaves me wondering if I'm missing something else.

Has anyone else run into this kind of issue with the Mercury API before?

Adrianrr94, 2019-01-31:
Hello Dan,

As you said, I tried to read the user memory with a network reader, obtaining as a result None everytime I try to acces the user memory. 

I was wondering if you could solve this problem, in case you do I would appreciate if you could share the solution.

Thanks.

DanWehr, 2019-02-20:
@Adrianrr94 you will be happy to know that I have a working solution to this now. I should be able to initiate a pull request for it before the end of the month once I find some free time.