gotthardp/python-mercuryapi #49: User Bank

rasidmusic, 2019-01-28:
Hey Guys,

I have achieved in reading epc asynchrouns.
In the How To i read that with the bank="user" parameter, I can read userdata that I have written to the tags.
But the following code just does nothing:
`import mercury
reader = mercury.Reader("tmr:///dev/ttyUSB0")

reader.set_region("EU3")
reader.set_read_plan([1], "GEN2", bank=["user"],read_power=500)
reader.start_reading(lambda tag: print(tag.user_mem_data))
`

Any ideas?

rasidmusic, 2019-01-28:
EDIT:
 I tried the same with teh reader.read() Function, but stil, I dont get the User data, but the EPCs:

`import mercury
reader = mercury.Reader("tmr:///dev/ttyUSB0")

reader.set_region("EU3")
reader.set_read_plan([1], "GEN2", bank=["user"],read_power=500)
print(reader.read())`

Result:
`import mercury
reader = mercury.Reader("tmr:///dev/ttyUSB0")

reader.set_region("EU3")
reader.set_read_plan([1], "GEN2", bank=["user"],read_power=500)
print(reader.read())`

florianhumblot, 2019-02-08:
As far as I know the `reader.start_reading()` method does not allow reading bank data (also see: https://github.com/gotthardp/python-mercuryapi/issues/39#issuecomment-442828272) 

Have you tried leaving the read_power in the `set_read_plan` function untouched? I'm using the exact same parameters except for the read_power, it might be that the connection is too feeble at 500 to extract the data.

My read plan is as follows:

`set_read_plan(rfid_reader.get_antennas(), "GEN2", bank=["user", "epc", "tid", "reserved"])`

followed by sequential read calls as follows: 

`tag_data = rfid_reader.read(timeout=450)`

callen5914, 2019-02-16:
I would love to know how in the hell you got it working!

I followed the usage guide and get the damn TypeError: Timeout

I get the light flashing on the FTDI cable. So I know the URI is right. Baudrate makes no difference.

`import mercury

reader = mercury.Reader("tmr:///dev/ttyUSB0", baudrate=9600)`

rasidmusic, 2019-02-19:
> As far as I know the `reader.start_reading()` method does not allow reading bank data (also see: [#39 (comment)](https://github.com/gotthardp/python-mercuryapi/issues/39#issuecomment-442828272))
> 
> Have you tried leaving the read_power in the `set_read_plan` function untouched? I'm using the exact same parameters except for the read_power, it might be that the connection is too feeble at 500 to extract the data.
> 
> My read plan is as follows:
> 
> `set_read_plan(rfid_reader.get_antennas(), "GEN2", bank=["user", "epc", "tid", "reserved"])`
> 
> followed by sequential read calls as follows:
> 
> `tag_data = rfid_reader.read(timeout=450)`

I have tried your method:
```
reader.set_read_plan([1], "GEN2", bank=["user"], read_power=1400)
allTags.extend(reader.read(timeout=450))
```
However, print returns the EPC
```
print(allTags)
>>> [b'EPCLONGNUMBERHERE]
```
Why doesnt it return the User data as specified in the set_read-plan?

rasidmusic, 2019-02-19:
> I would love to know how in the hell you got it working!
>
> I followed the usage guide and get the damn TypeError: Timeout
>
> I get the light flashing on the FTDI cable. So I know the URI is right. [...]

Give us some more insights on your code and the specific Timeout Error. Maybe the Reader is still active and a new connection causes the timeout.

florianhumblot, 2019-02-19:
> I have tried your method:
> 
> ```
> reader.set_read_plan([1], "GEN2", bank=["user"], read_power=1400)
> allTags.extend(reader.read(timeout=450))
> ```
> 
> However, print returns the EPC
> 
> ```
> print(allTags)
> >>> [b'EPCLONGNUMBERHERE]
> ```
> 
> Why doesnt it return the User data as specified in the set_read-plan?

You get an object back, whose representation is the EPC. To get the actual data from the banks you need to access them explicitly like so:

```python3
tag.epc.__str__()[1:], #This is to get the string representation of the EPC instead of a bytearray
tag.antenna,
tag.read_count,
tag.rssi,
str(tag.epc_mem_data),
str(tag.tid_mem_data),
str(tag.user_mem_data)
```

florianhumblot, 2019-02-19:
> I would love to know how in the hell you got it working!
>
> I followed the usage guide and get the damn TypeError: Timeout
>
> I get the light flashing on the FTDI cable. So I know the URI is right. [...]

Have you checked that it is the correct port? Mine connects to `ttyUSB1` for some reason. 
Also, have you tried without setting the baudrate?
Is the reader connected to the right USB port? My reader (m6e) has 2 mini-usb ports and only one functions as a USB device, the other one functions as some kind of serial interface which will not work.

florianhumblot, 2019-02-19:
> Hey, thanks for the How to. It works - but not entirely.
> I am able to get the user data with your code.
> But with my code, where I want to get the user data of all read tags, only the first one returns user data. Although I know that every tag has user data.
> 
> My Code:
> 
> ```
> reader.set_read_plan([1], "GEN2", "user", read_power=1400)
> all_tags_b = reader.read(1000)
> for i in all_tags_b:
>             print("Das ist ein Tag: ", i, i.user_mem_data, i.read_count)
> ```
> 
> The Result only returns the user data for the first detected tag(all tags have user data):
> 
> ```
> Das ist ein Tag:  b'E20000196302011223302804' bytearray(b'L16030108\r\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00') 10
> Das ist ein Tag:  b'E200001963020091231027DE' None 8
> Das ist ein Tag:  b'E200001963020073233027B9' None 6
> Das ist ein Tag:  b'E200001963020190233028A3' None 7
> Das ist ein Tag:  b'E2000019630201152330280A' None 8
> ```
> 
> Any Ideas?

I don't have access to the reader right now, but my best guess is that the antenna wasn't able to read the bank of all the tags.

Maybe they're too far away, or obstructed, but the way your code is written, it should definitely work.

Just in case, you can try using the Universal Reader Assistant from the makers of the api, I'm pretty sure it's either that the signal isn't strong enough or that it's a limit of the API although I highly doubt the latter.

rasidmusic, 2019-02-20:
The are all sticking onto some boxes and are about 10cm away from the reader...this should not be the problem.
The Universa Reader Assistant shows all user data.....

florianhumblot, 2019-02-20:
> 
> 
> The are all sticking onto some boxes and are about 10cm away from the reader...this should not be the problem.
> The Universa Reader Assistant shows all user data.....

I'm guessing that it's a bug in the library then (as in this particular python port) i'd open a different issue addressing that bug and close this one if I were you as to not confuse anyone and considering your issue has been solved :)

closed 2019-02-20 by rasidmusic