gotthardp/python-mercuryapi #52: BUG? Multiple Tags - User Data Read only on first Tag

rasidmusic, 2019-02-20:
Hey @gotthardp 
I tried the following:
```
#!/usr/bin/env python3
from __future__ import print_function
import mercury
reader = mercury.Reader("tmr://dev/ttyUSB0", baudrate=115200)

reader.set_region("EU3")
reader.set_read_plan(reader.get_antennas(), "GEN2", "user")
for tag in reader.read(timeout=450):
    print("EPC", tag.epc)
    print("Antenna", tag.antenna)
    print("Count", tag.read_count)
    print("RSSI", tag.rssi)
    print("user", tag.user_mem_data)
```
 Unfortunately I get the following results:
```
EPC b'E200001963020091231027DE'
Antenna 1
Count 7
RSSI -37
user bytearray(b'L18100117\r\n\x00\x00\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
EPC b'E2000017230C01961100AA8E'
Antenna 1
Count 7
RSSI -31
user None
```
As you can see, the second tag does not return user data. That is, if two tags are read at the same time. Both show the correct user data if read seperately.

_Originally posted by @rasidmusic in https://github.com/gotthardp/python-mercuryapi/issues/39#issuecomment-465491824_

florianhumblot, 2019-02-22:
Hey, I'm wondering if it could have to do with the information stored in the user bank.

Could you change your read plan to:
`set_read_plan(rfid_reader.get_antennas(), "GEN2", bank=["user", "epc", "tid"])`

And also printing the tag's epc and tid data?

rasidmusic, 2019-02-25:
Hey,
I've tried your code:
```
#!/usr/bin/env python3
from __future__ import print_function
import mercury
reader = mercury.Reader("tmr:///dev/ttyUSB0", baudrate=115200)

reader.set_region("EU3")
reader.set_read_plan(reader.get_antennas(), "GEN2", bank=["user", "epc", "tid"])
for tag in reader.read(timeout=450):
    print("EPC", tag.epc)
    print("Antenna", tag.antenna)
    print("Count", tag.read_count)
    print("RSSI", tag.rssi)
    print("user", tag.user_mem_data)
    print("tid", tag.tid_mem_data)
```

Result:

```
EPC b'E200001963020091231027DE'
Antenna 1
Count 6
RSSI -37
user bytearray(b'L18100117\r\n\x00\x00\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
tid bytearray(b'\xe2\x004\x12\x01<\xfd\x00\x0b\xbd&\x80\x10\x03\x0170\r_\xfb\xff\xff\xdcP')
EPC b'E2000017230C01961100AA8E'
Antenna 1
Count 5
RSSI -31
user None
tid None
```

So the TID of the second Tag is not read. It seems that xyz_mem_data is only read on the first detected tag...

rasidmusic, 2019-02-25:
Tried also with adding
`print("epc", tag.epc_mem_data)`

Same result, shows only EPC of the first detected tag

gotthardp, 2019-06-14:
Sorry, I cannot reproduce it. I tried your code with two tags with a single antenna, and the system correctly reads user data from both simultaneously.

closed 2019-06-14 by gotthardp