gotthardp/python-mercuryapi #54: pip install fails on Ubuntu 16.04

romainrobotics, 2019-03-06:
Attempting to install with

pip install python-mercury api

From here:
https://pypi.org/project/python-mercuryapi/

This is the output

Collecting python-mercuryapi
  Using cached https://files.pythonhosted.org/packages/5f/b3/0ed60331a77de7fcc5f8dc55bd48a901a327a93052e2756868cd9ead2ac2/python-mercuryapi-0.4.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-s0UN60/python-mercuryapi/setup.py", line 8, in <module>
        long_description=open('README.md').read(),
    IOError: [Errno 2] No such file or directory: 'README.md'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-s0UN60/python-mercuryapi/

When I build from source using python2, then try to import, it says it's already been imported and it just doesn't work at all.

gotthardp, 2019-03-08:
Sorry for that. It should work now.

closed 2019-03-08 by gotthardp

romainrobotics, 2019-03-11:
Awesome, thank you