gotthardp/python-mercuryapi #55: Bug start_reading

Greenerwave, 2019-03-07:
Hello,

I encouter an issu with the function start_reading with the folling code: 
```
import mercury
reader = mercury.Reader("tmr:///COM12")

    
''' paramètres'''
reader.set_read_powers([1, 2, 3, 4], [2100,2300,2200,2100])
reader.set_region("EU3")
reader.set_read_plan([1,2,3],"GEN2")
reader.set_gen2_session(0)
reader.set_gen2_q(0,2)

print("start")
reader.start_reading(lambda tag: print(tag.epc))
counts = 5
while counts:
    print('iteration', counts)
    counts = counts - 1
reader.stop_reading()
print("end")
```

I do not receive any error message however it forces us to restart the shell as shown in this image:
[image]

Do you have any idea how to fix it?

thank's

gotthardp, 2019-03-08:
I haven't seen this before, but I suspect it may be a memory access problem. Could you verify it using ` valgrind --trace-children=yes ./yourscript`, please?

Greenerwave, 2019-03-11:
thank you for your answer, I work under windows do you have another solution to offer me under windows?
Thank's

Greenerwave, 2019-03-15:
hello,
I went from windows to Linux to check the memory access error and I used the command valgrind, I have a lot of error so if you can help me find a solution, please.
Thank's

[image]
[image]

gotthardp, 2019-03-15:
Thanks for your help. Two questions:
* Did you get the crash also on Linux?
* In the list of errors that you see, is there any error that leads to the wrapper itself? I mean: at the end of each line there is a path (e.g. /usr/bin/python3.6). If you ignore all these "native" Python errors, is there any other error, which would leads to the python-mercuryapi sources?

Greenerwave, 2019-03-15:
thanks for answering quickly,
yes you're right I think all the mistakes come from Python, it's a bit weird I thought that I have it well installed
how can I install the crash,
sorry but I do not really know Linux commands.
thank you

Greenerwave, 2019-03-15:
I found this command:    sudo rm /var/crash/*
that's what i have to do?

Thank's

gotthardp, 2019-03-15:
I meant whether you are experiencing the restart (you reported above) also in Linux.

The Python errors are normal, just ignore them. There is nothing wrong with your installation; I get these errors everytime. The question is whether you see something else, some error other than the Python ones.

Greenerwave, 2019-03-18:
in fact when I increase the time sleep the reading continues and stops at a time but I do not have the display of reading stop! just the window crashes.
Thank's
[image]

Greenerwave, 2019-03-18:
Hello,
Thank you I solved the problem it comes from our inactive python.

gotthardp, 2019-03-18:
Cool

closed 2019-03-18 by gotthardp

SilverMecer, 2019-04-18:
> in fact when I increase the time sleep the reading continues and stops at a time but I do not have the display of reading stop! just the window crashes.
> Thank's
> [image]

Hi, I am having the same the issue, will you please so kind and share your solution to the issue?

Thank you

Greenerwave, 2019-05-02:
The  solution for me was to compile the programm under linux command, for
me the problem was with idle python3, so i changed the compilation method
with this command (python3 test.py).

Fatima-ezzahraa Majid
+33 7 69 14 37 02