gotthardp/python-mercuryapi #56: pip install fails on Raspberry Pi

romainrobotics, 2019-03-11:
I'm not sure how pip works in terms of installing dependencies before installing the target library, but some necessary dependencies are apparently not present in Raspbian. Simply running 

pip2 install python-mercuryapi 

will fail unless first running 

sudo apt-get install patch xsltproc gcc libreadline-dev python-dev python-setuptools

Like I wrote, I don't know how pip works but I thought it was possible to build in the installation of dependencies first, so maybe that could be added in.

gotthardp, 2019-03-11:
I don't think it is possible to specify OS dependencies in the pip file. But I will at least update the documentation. Thanks.

gotthardp, 2019-06-11:
Documented already.

closed 2019-06-11 by gotthardp