gotthardp/python-mercuryapi #58: reader.start_reading Segmentation Fault

mitchgalea, 2019-03-14:
Hello,

I'm trying to use the reader.start_reading() function with ROS (Robot Operating System) on Linux. When a reading is taken I wish to add the tag to a Queue variable, readings. However whenever a tag is read it seg faults. I come from a c++ background and am not too familiar with python a c integration as done within this library. I suspect that it is seg faulting when trying to access the readings variable from within the mercury api. 
Is this possible to fix or will I have to do a work around with the reader.read() function? 
`

	import rospy
	import mercury

	from tm_rfid_reader.msg import RFIDReading
	from multiprocessing import Queue

	def talker():
		pub = rospy.Publisher('rfid_reading', RFIDReading, queue_size=10)
		reader = mercury.Reader("tmr:///dev/ttyUSB0")
		print(reader.get_supported_regions())
		print(reader.get_power_range())
		print(reader.get_antennas())
		print(reader.get_model())

		reader.set_region("AU")
		reader.set_read_powers([1], [1000])
		reader.set_read_plan([1], "GEN2")
		print(reader.get_read_powers())

		rospy.init_node('rfid_reading_pub', anonymous=True)

		readings = Queue()
		reader.start_reading(lambda tag: readings.put(tag))

		rate = rospy.Rate(10)
		while not rospy.is_shutdown():
		    tag = readings.get()
		    reading = RFIDReading(tag.epc, tag.read_count, tag.rssi, 0)
		    pub.publish(reading)
		    rate.sleep()

	if __name__ == '__main__':
		try:
		    talker()
		except rospy.ROSInterruptException:
		    pass
`

Adrianrr94, 2019-03-18:
Hi mitchgalea,

I faced the same problem, I think that the segmentation fault happens when the buffer memory's reader is full. 
So what I made is the following:
`

while read_from_sensor:
       reader.start_reading(lambda tag: values(tag.epc, tag.antenna, tag.read_count, tag.rssi))
       time.sleep(1)
       reader.stop_reading()

`
I don't know if it is the best way to solve this problem, but it is the only thing that works for me.
The variable read_from sensor will be false when the operation is done.

I hope this works mate.

mitchgalea, 2019-03-18:
Cheers mate, 
That worked

closed 2019-03-18 by mitchgalea