gotthardp/python-mercuryapi #59: Can't use any getters

florianhumblot, 2019-03-19:
Hi there,

I'm trying to use the different functions to get more information about the way my reader is setup.
However, when I try to call the functions, the `mercury.Reader` object spits out an `AttributeError`.

Hardware: ThingMagic M6E
OS: Windows 10
Python: 3.6
Build: Latest master

Program:

```python3

import mercury

def print_tag(tag):
    print(tag)

if __name__ == '__main__':
    reader = mercury.Reader("tmr:///COM6")
    reader.set_read_plan(reader.get_antennas(), "GEN2")
   
```

to whitch  I add either of these lines:
```python3
print(reader.get_gen2_blf())
print(reader.get_gen2_tari())
print(reader.get_gen2_tagencoding())
print(reader.get_gen2_session())
print(reader.get_gen2_target())
print(reader.get_gen2_q())
```

The specific error I get is:

`AttributeError: 'mercury.Reader' object has no attribute 'get_gen2_blf'` 

where the attribute changes with the command used.

The other get commands work, although they sometimes give weird results, such as:

``` python3
print(reader.get_power_range())

(1813119476, 1813122126)
```

Is this a known issue or a bug? 

Thanks!

gotthardp, 2019-03-19:
Hello. Could you please double check this is not a build/installation issue? Could you double check there is no other code installed somewhere, which could be called or mislinked?

florianhumblot, 2019-03-19:
Hi @gotthardp , I've double checked and I have fully uninstalled the previous version when updating, I can't find any remnants from a previous installation.

florianhumblot, 2019-03-19:
@gotthardp apparently the getters aren't defined in mercury.py, could that be the issue?

gotthardp, 2019-03-20:
There is no `mercury.py`, is there?

florianhumblot, 2019-03-20:
There is, it seems to be auto-generated when installing, when I use my IDE's look-up feature (ctrl+click on a function in clion for example) it opens a file called mercury.py which does not contain the getter interfaces.

This is a stub file that my IDE shows:

```python3

# encoding: utf-8
# module mercury
# from C:\Python36\lib\site-packages\mercury.cp36-win32.pyd
# by generator 1.146
""" ThingMagic Mercury API to control ThingMagic RFID readers. """
# no imports

# no functions
# classes

class Reader(object):
    """ Reader object """
    def get_antennas(self, *args, **kwargs): # real signature unknown
        """ Lists available antennas. """
        pass

    def get_model(self, *args, **kwargs): # real signature unknown
        """ Returns the model name """
        pass

    def get_power_range(self, *args, **kwargs): # real signature unknown
        """ Lists supported radio power range. """
        pass

    def get_read_powers(self, *args, **kwargs): # real signature unknown
        """ Lists configured read powers for each antenna. """
        pass

    def get_supported_regions(self, *args, **kwargs): # real signature unknown
        """ Returns a list of regions supported by the reader """
        pass

    def get_temperature(self, *args, **kwargs): # real signature unknown
        """ Returns the chip temperature """
        pass

    def read(self, *args, **kwargs): # real signature unknown
        """ Read the tags """
        pass

    def set_read_plan(self, *args, **kwargs): # real signature unknown
        """ Set the read plan """
        pass

    def set_read_powers(self, *args, **kwargs): # real signature unknown
        """ Set the read power for each listed antenna and return the real setted values. """
        pass

    def set_region(self, *args, **kwargs): # real signature unknown
        """ Set the reader region """
        pass

    def start_reading(self, *args, **kwargs): # real signature unknown
        """ Start reading tags asynchronously """
        pass

    def stop_reading(self, *args, **kwargs): # real signature unknown
        """ Stop asynchronous reading """
        pass

    def write(self, *args, **kwargs): # real signature unknown
        """ Write the epc_target tag with the given epc_code """
        pass

    def __init__(self, *args, **kwargs): # real signature unknown
        pass

    @staticmethod # known case of __new__
    def __new__(*args, **kwargs): # real signature unknown
        """ Create and return a new object.  See help(type) for accurate signature. """
        pass

class TagData(object):
    """ TagData object """
    def __init__(self, *args, **kwargs): # real signature unknown
        pass

    def __repr__(self, *args, **kwargs): # real signature unknown
        """ Return repr(self). """
        pass

    epc = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default
    """Tag EPC"""

class TagReadData(TagData):
    """ TagReadData object """
    def __init__(self, *args, **kwargs): # real signature unknown
        pass

    antenna = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default
    """Antenna where the tag was read."""

    epc_mem_data = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default
    """EPC bank data bytes"""

    read_count = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default
    """Number of times the tag was read."""

    reserved_mem_data = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default
    """Reserved bank data bytes"""

    rssi = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default
    """Strength of the signal recieved from the tag."""

    tid_mem_data = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default
    """TID bank data bytes"""

    user_mem_data = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default
    """User bank data bytes"""

# variables with complex values

__loader__ = None # (!) real value is ''

__spec__ = None # (!) real value is ''

```

gotthardp, 2019-03-21:
This is weird. Could it be a build problem?

florianhumblot, 2019-03-22:
hmm, I think I've figured it out.
It's working now on a debian based system, I think that it went wrong because the installation didn't update the virtual environment, you need to use the system interpreter.

My bad

closed 2019-03-22 by florianhumblot