gotthardp/python-mercuryapi #61: `get_read_powers` returns incorrect results

RaphiePS, 2019-04-22:
I'm calling `reader.get_read_powers()` and it returns: `[(196, 2128200136)]`. I wish I had 196 antennas and that much power, but alas, it appears to be incorrect.

python-mercuryapi commit: `a264834e3d7840b511b4cc1612a40084236c356b` (latest)
Mercury version: `mercuryapi-1.31.2.40`
Computer: Raspberry Pi
Reader: SparkFun M6E Nano breakout 

Apart from that, everything is working great. Thanks for the stellar library!

RaphiePS, 2019-04-23:
For what it's worth, `reader.get_power_range()` is returning `(1989410816, 1989413516)`.

gotthardp, 2019-04-23:
And what is `reader.get_antennas()` returning, please?

RaphiePS, 2019-04-23:
`reader.get_antennas()` returns `[1]`. Here's the full output of me configuring the Nano (though removing the configuration still gives the faulty power readings).
```
ic| reader.get_antennas(): [1]
ic| reader.get_model(): 'M6e Nano'
ic| reader.set_region("NA2"): None
ic| reader.set_gen2_tari(0): 0
ic| reader.set_gen2_blf(250): 250
ic| reader.set_gen2_tagencoding(3): 3
ic| reader.set_gen2_session(0): 0
ic| reader.set_gen2_target(2): 2
ic| reader.set_gen2_q(0, 4): (0, 4)
ic| reader.set_read_powers([1], [POWER]): [(1, 2000)]
ic| reader.get_power_range(): (1989738496, 1989741196)
```

Jack839, 2019-06-08:
Any solution yet?

gotthardp, 2019-06-11:
I forgot about this, sorry.

I made some update that could fix this issue. Would you please be able to download the latest HEAD and try the fix, please?

Jack839, 2019-06-11:
I will try the fix and tell you soon.

Jack839, 2019-06-12:
Sir it is still showing wrong values. Although get_power_range is also giving [500, 3150] with m6e with 4 antenna which is quite accurate than previous values.

gotthardp, 2019-06-12:
So what does the `get_antennas`, `get_read_powers` and `get_power_range` return now, please?

Jack839, 2019-06-12:
M6E with 4 antennas

get_antennas gives [1, 2, 3, 4]
get_read_powers gives [(1, 0), (8, 32530), (104, 32530), (64, 0)]
get_power_range gives (500, 3150)

gotthardp, 2019-06-12:
and if you `set_read_powers` to something, the value persists and the same is returned in `get_read_powers` or not?

Jack839, 2019-06-13:
after setting reader.set_read_powers([1], [3000]) following values were given :
get_antennas gives [1, 2, 3, 4]
get_read_powers gives [(1, 3000), (8, 32624), (104, 32624), (64, 0)]
get_power_range gives (500, 3150)

gotthardp, 2019-06-13:
I believe I fixed this now.

Please note that:
 * the list returned from `get_read_powers` does not include antennas with default power setting, so the list may be empty-- this was actually the problem.
 * I modified the syntax of `set_read_powers` to a list of tuples (same as in `get_read_powers`)

Jack839, 2019-06-13:
Okay Sir, I will try this fix and tell the result after merging this commit with existing mercury library.

Jack839, 2019-06-14:
It working as per your told.

closed 2019-06-14 by gotthardp