gotthardp/python-mercuryapi #62: sleep parametres in asynchronous reading

Greenerwave, 2019-05-21:
hello!
I want to know how to set sleep characteristic times in asynchronous reads,

THANKS

gotthardp, 2019-05-21:
Hi. You mean on/off times? It's a parameter of start_reading.
https://github.com/gotthardp/python-mercuryapi#readerstart_readingcallback-on_time250-off_time0

Greenerwave, 2019-05-21:
No 
i mean the sleep between start reading and stop reading

Greenerwave, 2019-05-21:
reader.start_reading(lambda tag : acquisition(tag))
        tm.sleep(10)--------- this line
        reader.stop_reading()

gotthardp, 2019-05-21:
Well, that's just an example. For asynchronous operations you start reading and the **do something else**. In this case I just wait for few secs, but in reality you can stay there forever.

Greenerwave, 2019-05-22:
Thanks
Okey , when I take off the sleep for the code, the start.reading sent no data, do you have any idea about this problem?

gotthardp, 2019-05-22:
Start_reading is not supposed to send anything. In the time between start_reading and stop_reading the reader reads RFID tags and invokes the given lambda fun.
How does your entire code look like?

Greenerwave, 2019-05-22:
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
from __future__ import print_function
from time import sleep
import time
import mercury
reader = mercury.Reader("tmr:///dev/ttyUSB0", baudrate=115200)

start_time = time.time()

reader.set_read_powers([1,2,3,4], [2100,2300,2200,2100])
reader.set_region("EU3")
reader.set_read_plan([2], "GEN2", read_power=2100)

print(reader.get_supported_regions())
print(reader.get_power_range())
print(reader.get_antennas())
print(reader.get_read_powers())

taglist=[]
taglist2=[]

print("start")

def printtag(tag):
    taglist.append(tag.epc)
    taglist2.append(tag.rssi)

reader.start_reading(printtag, on_time=1000,  off_time=0)

#time.sleep(1)
#reader.stop_reading()

print(taglist)
print(taglist2)
print("end")

print("Temps d'éxecution : %s secondes ---" % (time.time() - start_time))

when I take off the sleep for the code, the start.reading sent no data,!!

Result: 

start
[ ]            -------empty list
[ ]            -------empty list
end
Temps d'éxecution : 0.259052038193 secondes ---

THANKS

gotthardp, 2019-05-22:
Well, you start reading and terminate the program execution short after that. The reader has no time to read anything. I wonder-- why are you using start_reading if your code doesn't need any asynch behaviour? Wouldn't plain read be more suitable in this case?

Greenerwave, 2019-05-23:
yes I agree but I will use asynchronous for the continuation of the code, it's okey  I solved the problem using the multithreading

THANKS

closed 2019-05-23 by Greenerwave