gotthardp/python-mercuryapi #65: Segmentation Fault on long asynchronous reading (start_reading())

Jack839, 2019-06-13:
I have found a new issue which actually give segmentation fault on long asynchronous reading.
After reading continously around 60000 times in total with may tags it returns segmentation fault.
I have attached a image which was the problem that api gave on debugginging with gdb.
[image]

gotthardp, 2019-06-13:
Could you try displaying `bt` in gdb? This is too little information to debug something.

Jack839, 2019-06-13:
I will show it you soon.

Jack839, 2019-06-14:
backtraced: 
[image]

gotthardp, 2019-06-14:
I made an update that could possibly fix the error. Could you please try and let me know if it helped?

Jack839, 2019-06-14:
Will try and tell you.

Jack839, 2019-06-15:
Looks like this is solved. I read around 1.2 lakhs times without any issue.

closed 2019-06-15 by Jack839