gotthardp/python-mercuryapi #66: GPIO Pins implementation

Jack839, 2019-06-13:
Is there any way to get the gpio pins working for the reader? ☹️
Because I really need them for a project. If you can do it, it would be very nice or else you might teach me or tell me the resources to learn how to implement c api features for python?

Thank you sir.

gotthardp, 2019-06-13:
Duplicate of #33.

I will implement it.

Jack839, 2019-06-13:
Okay Sir, thank you .

closed 2019-06-13 by Jack839

gotthardp, 2019-06-13:
Done. (I am extremely fast today :) )

Jack839, 2019-06-13:
Yipee....thank you sir

Jack839, 2019-06-13:
Can you still teach me how to implement any new functionality in the library?

gotthardp, 2019-06-13:
Have a look how the GPIO is done. It's in one quite clean commit:
https://github.com/gotthardp/python-mercuryapi/commit/58e28774844a8aae1543bcbab452ea41627dd9c8

Jack839, 2019-06-14:
Okay sir, I will check it out.