gotthardp/python-mercuryapi #67: reader.gpo_set(2, False) is not working.

Jack839, 2019-06-14:
Here is the error:
[image]

gotthardp, 2019-06-14:
Does `reader.get_gpio_outputs()` include the pin you want to set?

Jack839, 2019-06-14:
When this code was ran:
`print("GPIO Inputs:", reader.get_gpio_inputs())
print("GPIO Outputs:", reader.get_gpio_outputs())
#reader.gpo_set(2, False)
print("GPI GET:", reader.gpi_get(2))
reader.set_gpio_inputs([1, 2])
print("GPIO Inputs:", reader.get_gpio_inputs())
reader.set_gpio_outputs([3, 4])
print("GPIO Outputs:",reader.get_gpio_outputs())`
Output:
[image]

gotthardp, 2019-06-14:
Looks good, or? The gpo_set shall be called **after** you clear the pin from gpio_inputs.

Jack839, 2019-06-14:
Will try and tell you.

Jack839, 2019-06-15:
It is working fine after doing that but is it supposed to stay stuck at that line for keeping the gpio pin high? Because code is stuck at that line and not proceeding further.

gotthardp, 2019-06-16:
What is the call-stack when you interrupt the script by ctrl-c?

Jack839, 2019-06-18:
It is working fine sir. It was my mistake.

closed 2019-06-18 by Jack839