gotthardp/python-mercuryapi #68: Writing data to tags

NicolasE124, 2019-06-19:
Hello,

I would like to write data on my tags. From what I see the reader.write function only allows to change the EPC. Is there a way to write on a different bank of the tag or is it not implemented in the wrapper yet?
Kind regards

gotthardp, 2019-06-19:
Hello. I am afraid that only the EPC can be written at this moment.

NicolasE124, 2019-06-19:
Thanks for your quick answer! Although it is not the one I was hoping for :)
Is there a way to implement it myself (knowing that I am not really an expert in C)? If you have any doc on the method used for other functions that would be much appreciated. In the meantime I will have a closer look at your code see if I can get some understanding.
Regards

gotthardp, 2019-06-19:
What exactly you want to write to the tag?

NicolasE124, 2019-06-19:
So, I am trying to play with some NXP RFID microchip which has 2 RF ports (https://www.nxp.com/docs/en/data-sheet/SL3S4011_4021.pdf). From what I can see there is a specific place in the EPC bank which allows to select which RF port is active. I would like to be able to modify that. Thanks again

gotthardp, 2019-06-19:
I've added `read_tag_mem` and `write_tag_mem` functions, which is probably what you need.

NicolasE124, 2019-06-19:
That is awesome, I'll give it a go asap
Thanks a lot for the support!

closed 2019-06-20 by NicolasE124

gotthardp, 2019-06-20:
I am glad it works :)

Bo-Bo-Han, 2023-10-16:
> Hello. I am afraid that only the EPC can be written at this moment.