gotthardp/python-mercuryapi #69: Input/Output error after leaving USB M5e-C reader plugged in for prolonged time

CatsAreCrits, 2019-06-19:
Hi, thank you so much for this wrapper and your support!

I am using the M5e-C reader plugged into a Windows 10 machine. I have a program that will connect to the reader and return the scanned tag when the user presses CTRL+M. It runs within a tkinter GUI loop.

This works fine when you first plug in the reader. But after a prolonged time (not sure exactly how long, at least a few hours), it will stop working and return this error.

`Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Users\<redacted>\AppData\Local\Programs\Python\Python36-32\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\<redacted>\AppData\Local\Programs\Python\Python36-32\lib\threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\<redacted>\AppData\Local\Programs\Python\Python36-32\lib\site-packages\keyboard\_generic.py", line 58, in process
    if self.pre_process_event(event):
  File "C:\Users\<redacted>\AppData\Local\Programs\Python\Python36-32\lib\site-packages\keyboard\__init__.py", line 216, in pre_process_event
    callback(event)
  File "C:\Users\<redacted>\AppData\Local\Programs\Python\Python36-32\lib\site-packages\keyboard\__init__.py", line 647, in <lambda>
    handler = lambda e: (event_type == KEY_DOWN and e.event_type == KEY_UP and e.scan_code in _logically_pressed_keys) or (event_type == e.event_type and callback())
  File "\\server\Repositories\<redacted>\production_thingmagickeyboard.pyw", line 77, in scan_tag
    start_reader(the_string)
  File "\\server\Repositories\<redacted>\production_thingmagickeyboard.pyw", line 42, in start_reader
    reader = mercury.Reader("tmr:///com3", baudrate=9600)
TypeError: Input/output error`

Here is the start_reader():
`def start_reader(the_string):
			# start up the reader
			reader = mercury.Reader("tmr:///com3", baudrate=9600)
			# required pre-requisite setup
			reader.set_region("NA")
			reader.set_read_plan([1], "GEN2")
			# this power is the equivalent of 12 dBm
			# this power gives the device a read range of <6 inches
			reader.set_read_powers([1], [1200])
			# read the tag(s), for each tag read type it out on a keyboard then press enter
			reads = reader.read()
			if len(reads) > 1:
				keyboard.write("error1")
				the_string.set(str("WARNING: Too many labels scanned!"))
			elif len(reads) == 0:
				keyboard.write("error0")
				the_string.set(str("WARNING: NO labels scanned!"))
			else:
				for read in reads:
					tag = read.epc.decode("utf-8")
					keyboard.write(tag)
					print(tag)
					the_string.set(str("Successfully scanned: " + tag))
			the_string.set(str("Typing it out now.."))`

Do you have any ideas? I tried looking through the mercury documentation but could not find this error. [Also I don't think it's related to this issue](https://github.com/gotthardp/python-mercuryapi/issues/41).

I also tried disabling "USB selective suspend setting" in windows, but this did not fix it either.

Please let me know if you have any ideas. Thank you!

gotthardp, 2019-06-19:
Hello. The issues that occur occasionally are hard to debug.

Do you have to initialize the reader every time? Could you try initializing the reader only once and then only calling `reader.read()` upon button press?

As a general experience, repeated termination and re-initialization may cause the reader some headache (not sure why).

CatsAreCrits, 2019-06-19:
That was how I initially did it and I had the same problem. I used to initiate it with the GUI and then call reader.read() on button press. This issue is why I switched to this method, I should have stated that.

It doesn't occur occasionally, it occurs every single time after a prolonged period. It will work at first, but if I come back later and try it it is sure to fail. I've tried restarting my tkinter program but the only thing that fixes it is unplugging/replugging in the reader or rebooting windows.

gotthardp, 2019-06-19:
I see. (Stupid idea, but) have you tried some keepalive call? For example, initialize the reader on the beginning and then regularily invoke e.g. `get_temperature` to keep the system up?

CodyMorton, 2019-06-19:
I have not, I will try that now and run it today and see if the issue still persists. Thanks for the idea I hadn't thought of that!!

CodyMorton, 2019-06-19:
Just realized I'll have to revert to the old way of keeping it alive with the GUI, which is fine, but I'm pretty sure the issue still occurs even if the GUI isn't running. I hadn't realized that before either. Perhaps this is a issue with windows or the device, then.

CodyMorton, 2019-06-20:
The keepalive call did fix the issue. I ran `get_temperature` every 15 seconds. However the GUI/program had to be running permanently, which is unfeasible for my situation. To resolve this I am just changing the business procedure.

Also I think it should be noted for future reference:
The same error: `TypeError: Input/output error` occurs whenever you have a script running that is connected to the reader, then open another script that attempts to connect to the same reader. 

Thanks for your help. I think this is a reader or Windows issue.

closed 2019-06-20 by CodyMorton

gotthardp, 2019-06-20:
I see.

BTW, I just found a `reboot()` function. Not sure if it would help, but it's another option.