gotthardp/python-mercuryapi #70: Reader_gpo_set not working in 0.5.0 for Python 2.x

garyk10, 2019-06-19:
I just tried the latest build on a Pi 3 using Python 2.7.13.  Everything is working except for gpo_set.  During the make, this warning message printed out:

mercury.c:606:1: warning: ‘Reader_gpo_set’ defined but not used [-Wunused-function]
 Reader_gpo_set(Reader *self, PyObject *args)
 ^~~~~~~~~~~~~~

On lines 1385-1389 of mercury.c, the gpo_set function is only set for PY_MAJOR_VERSION >= 3.  I tried commenting out the IF statement on line 1385, and it remade okay.  But in python I get this error:
>>> reader.gpo_set(1, True)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: argument 2 (impossible<bad format char>)

Is there a reason that gpo_set is not working for Python 2.x?  Thank you!

gotthardp, 2019-06-19:
Yeah. Still somebody using 2.x? :) The reason is the Boolean attribute and `PyArg_ParseTuple` because it seems bool args `p` are supported since 3.3, but I am sure there is a workaround. I was just too lazy finding it :)

garyk10, 2019-06-19:
LOL!  Yes, 2.x.  A few small libraries are not ported yet, so all of our group's projects are in 2.x.

Thanks for the explanation!  I got it working on Python 2.7, including with Python's boolean conversion.  In mercury.c, I changed line 613 to:
    if (!PyArg_ParseTuple(args, "BB", &pin, &value))

And then removed the #if PY_MAJOR_VERSION >= 3 condition by deleting lines 1385 and 1389.

gotthardp, 2019-06-20:
I just fixed this in `master`. Thanks!

closed 2019-06-20 by gotthardp