gotthardp/python-mercuryapi #71: Temperature reading while asynchronous reading of tag not allowed.

Jack839, 2019-06-20:
Current I am trying to read temp and tag reading simultaneously but doing that gives error as "error getting temperature". Is it possible somehow? I think this feature is there in the c api.

gotthardp, 2019-06-20:
I just made an update that will improve the error report. It will not solve your problem, but the error report will be more specific and perhaps it will clarify what is wrong.

Jack839, 2019-06-20:
Now it shows this error: `RuntimeError: Unimplemented feature`

gotthardp, 2019-06-20:
This is coming directly from your reader, so it does not support it.

Jack839, 2019-06-20:
okay sir.

closed 2019-06-20 by Jack839

gotthardp, 2019-11-11:
New function `enable_stats` was just introduced to address this. https://github.com/gotthardp/python-mercuryapi/commit/7f493198056824cba9f8b11e89eadf46c4720de2