gotthardp/python-mercuryapi #73: Tagfilter

NicolasE124, 2019-06-28:
Hello,

Is it possible to filter on a specific tag when conducting a read operation? I would like to be able to focus on a given tag and get its RSSI as it is moving in the middle of other tags. I saw in the Mercury API Programmer guide there is such a fonction with the TagFilter interface and gen2.select function. But I don't see it yet in the wrapper. Is it something that could be done easily? Best regards

gotthardp, 2019-06-28:
Hello. Yes, it can be implemented.

gotthardp, 2019-07-02:
Implemented, but not tested. Please have a look and let me know if it works.

NicolasE124, 2019-07-03:
Hi,

Thanks again for your help. I am having some troubles with it this time.
I am trying with two tags only. I try to read them as usual once, then change the read plan to focus on the first one. Here is my code:

`import mercury`

`readers=mercury.Reader('tmr:///dev/ttyUSB0')`
`readers.set_read_plan([1],"GEN2")`
`tag=(readers.read())`
`print(tag)`
`readers.set_read_plan([1], "GEN2",epc_target=tag[0].epc)`
`print(readers.read())`

And the output:
 

> ['30393620006F6F800000161A', '30395DFA831CEA80000D2C6B']
> 
> Traceback (most recent call last):
>   File "/home/user/Documents/TestTarget/read1tagSync.py", line 8, in <module>
>     print(readers.read())
> RuntimeError: Message command length is incorrect

Sometimes it also gives:

> ['30393620006F6F800000161A', '30395DFA831CEA80000D2C6B']
> []

Do you think I am using the right argument for epc_target? Could it be a problem of encoding?
I will try a few different syntaxes and let you know if I get something new

Best regards

gotthardp, 2019-07-03:
There was an ugly incompatibility (the filter was expected as Unicode hexa-string). Now, its any string or bytes, so doing `epc_target=tag[0].epc` should be possible. Please try again.

NicolasE124, 2019-07-03:
Hi 

Well I still have the same messages actually but I am using Python2.7.
However, I then decided to switch to Python3.6 and it works perfectly! (maybe it worked with the previous version too).
For me it will be ok. I don't know if you want to find the issue in Python2.7 (I can close the issue if you want).
So thanks a lot for your help and reactivity again. If you come to Paris I should definitely offer you a drink!
Best regards

NicolasE124, 2019-07-03:
Works for both! Well done and thanks again

closed 2019-07-03 by NicolasE124