gotthardp/python-mercuryapi #74: Access readers via mac adress

rasidmusic, 2019-07-01:
Hey guys - hope you can help me out.
I am trying to use two readers at the same time.
I have usually assigned a reader with
`reader = mercury.Reader("tmr:///dev/ttyUSB0")`

With a second reader:
`readerA = mercury.Reader("tmr:///dev/ttyUSB0")`
`readerB = mercury.Reader("tmr:///dev/ttyUSB1")`

The problem: The ttyUSBx keep swapping around after a restart of the system. I need an exact way to use a defined reader. Is there a way of doing that via the mac address (which is unique of course)?
Something like
`readerA = mercury.Reader("23:df:34:ds:xy")`
`readerB = mercury.Reader("54:96:gt:le:43")`

Thank you

gotthardp, 2019-07-01:
I assume you are on Linux. Did you try defining udev rules to fix the device name? https://rolfblijleven.blogspot.com/2015/02/howto-persistent-device-names-on.html

If it's an IPv6 capable network (LAN) reader (and you access it via LAN/Ethernet) you may be able to access it via auto-generated local address, with is based on the device MAC.

rasidmusic, 2019-07-01:
Thanks for the link.
I tried to find an unique identifier to determine the exact reader - but could not find any.
idProduct and idVendor are the same (CH-340 both) and I can't find any attribute that is unique and persistent.
Any ideas?

gotthardp, 2019-07-01:
If there is no serial number? The article says there could be one (but not necessarily).

rasidmusic, 2019-07-01:
Apparently there is no serial number...

gotthardp, 2019-07-01:
Too bad. You will need to invent some trick.

One idea: There is a serial number in the MercuryAPI, so (if that works) you could initialize both readers and distinguish which is which afterwards using that number.
Another idea: Connect one using another method than the other (e.g. USB and LAN).

rasidmusic, 2019-07-01:
Note: I am using the Sparkfun RFID UHF Reader (based on the m6e nano) and a CH340 to connect it to the RasPi. Apparently the CH340 does not have any storage to store any type of unique id, like serial number (https://arduino.stackexchange.com/questions/6617/setting-serial-number-on-ch340-usb-serial-device). So  I need an FTDI

gotthardp, 2019-07-01:
The serial number I was talking about is stored inside the m6e nano. I added a new function `get_serial` that will (hopefully) print an unique string for each of the readers. Please check.

closed 2019-08-06 by gotthardp