gotthardp/python-mercuryapi #75: pip install fails with error code 1

andres-baller, 2019-07-03:
Attempting to install through pip3 on both macOS and Raspberry Pi 3B results in

> ERROR: Complete output from command python setup.py egg_info:
    ERROR: Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/87/_xg5mths3g35kb5x3kz6ddc80000gn/T/pip-install-mBTkEa/python-mercuryapi/setup.py", line 16, in <module>
        long_description=open('long_description.txt').read(),
    IOError: [Errno 2] No such file or directory: 'long_description.txt'
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/87/_xg5mths3g35kb5x3kz6ddc80000gn/T/pip-install-mBTkEa/python-mercuryapi/

I've already made sure setuptools is installed and updated, and other pip packages seem to be installing fine. The "no such file or directory" line makes me think it could be something similar to #54?

gotthardp, 2019-07-04:
Yeah, same as #54, just a different file. My fault. Will you try again now, please?

andres-baller, 2019-07-05:
It works on my Raspberry Pi now. It still doesn't build on my mac, but it's not crucial for me, and I'm assuming it's because of unrelated issues. Thanks for the quick support!

closed 2019-07-05 by andres-baller

gotthardp, 2019-07-05:
Yeah, the build on Mac doesn't work (see issue #42).