gotthardp/python-mercuryapi #77: write_tag_mem function not working properly

ACB2231, 2019-07-18:
Hello,

I am trying to use the write_tag_mem function but it only successfully writes when the user memory bank is specified, not the EPC bank. 

It also does not return true or false and when I use it the rest of my program does not run beyond the function, although no error is returned. 

Thank you for your help.

gotthardp, 2019-07-18:
Hello. What tag (exactly) are you using? What parameters you typed in (e.g. what bank number and address you set)? Did the command return something or Python just crashed?

ACB2231, 2019-07-18:
I am using a SL3S4021 tag, the bank number is 1 (EPC) and the address is 0x20.

The command does not return anything but python also does not crash, the program simply finishes without any further output. 

When I use the function on the 3 (user) memory bank, the same happens except when I read the same address afterwards it has been written to (this does not happen when writing to the EPC bank though).

gotthardp, 2019-07-18:
Forgive my stupid error, you print the result by `print` or not? The `print` prints nothing, an empty line or what?

ACB2231, 2019-07-18:
I do use print and it prints nothing at all.

gotthardp, 2019-07-18:
And the processing continues with the next Python command, or terminates (without any message)? (If this was the last command, would you mind adding some `print` after that?)

ACB2231, 2019-07-18:
The processing terminates after that even though I have already added some print commands after (any commands that come after the write in the program are not executed).

gotthardp, 2019-07-18:
That explains why the `print` does not print any result. How large data are you writing?

gotthardp, 2019-07-18:
One more question: are you using it with or without the `epc_target` filter?

ACB2231, 2019-07-18:
I am writing a byte array consisting of 16 two digit hexadecimal numbers and I am using it with the epc_target filter.

gotthardp, 2019-07-18:
I found a bug. Could you build the latest master and try it again, please?

ACB2231, 2019-07-19:
Hello, I have built the latest master but I still have the same problems

ACB2231, 2019-07-19:
Thank you, that has fixed the error; the function now returns whether it was successful and the subsequent lines in the program are now executed. 

When I try to write anything other than a bytearray of zeros to the EPC (1) memory bank I get the error "RuntimeError: Tag write operation failed" what could be causing this?

Thank you for your help.

gotthardp, 2019-07-19:
I don't know; the error comes directly from the reader. Maybe writing is prohibited (by the tag) or you need to write in smaller chunks.

closed 2019-08-06 by gotthardp