gotthardp/python-mercuryapi #82: Question : Targetting several tags

PierreAuriau, 2019-08-26:
Hello,

Is it possible to target more than one tag with the function set_read_plan ?

Thank you,

Pierre

gotthardp, 2019-08-26:
Hello. It is not possible at this moment, unfortunately.

PierreAuriau, 2019-08-27:
Ok, thank you very much !

gotthardp, 2019-08-27:
I will keep this issue open until it is fixed. (No date promised.)

gotthardp, 2019-08-28:
Question: Does somebody need this, or it it just a nice to have feature?

PierreAuriau, 2019-08-29:
It could be nice to have the feature because targetting a tag is much more efficient and faster than reading all tags and searching for the good ones !

gotthardp, 2019-08-29:
Hello again. I have some good and some bad news. I have implemented filtering of multiple tags, but on my reader this operation is not supported. You can give it a try; perhaps your reader is more sophisticated and will support it.

gotthardp, 2019-08-30:
Hello again. I finally understood and implemented the multi-filters, but it's a bit complicated, so I am not sure if I got the API right. Have a look and let me know what you think.

PierreAuriau, 2019-09-02:
Thank you very much for your work and sorry to answer lately.
I achieved to create a filter and to apply it with the function set_read_plan() but after that, the function reader.read() does not work anymore (with the error "Unsupported operation").
Do you know where this error comes from ? Shall I use asynchronous reading ?

gotthardp, 2019-09-02:
Hello. This used to be a problem of one of the previous commits. Please get the very latest code (comited few secs ago), build it and don't forget to install it. The error shall disappear.

PierreAuriau, 2019-09-02:
Re, I reinstall twice the wrapper but I still have the same issue. I may make a mistake, here my two-lign code : 
```python
epc_targets = [{'epc':b'1', 'bit':32}, {'epc':b'2', 'bit':32}]
reader.set_read_plan([1], 'GEN2',epc_targets, bank = ['epc'], read_power = 2900 )
reader.read()
```
Thank you !

gotthardp, 2019-09-02:
This `b'1'` and `b'2'` is what you really use?

PierreAuriau, 2019-09-03:
No no it was just an example in order to make my code clearer

gotthardp, 2019-09-03:
My code is
```python
reader.set_read_plan(reader.get_antennas(), "GEN2", bank=["epc"], epc_target=[{'epc':b'000000000000221612000631'}, {'epc':b'000000000000228510000023'}])
```
or even
```python
reader.set_read_plan(reader.get_antennas(), "GEN2", bank=["epc"], epc_target=[b'000000000000221612000631',b'000000000000228510000023'])
```

PierreAuriau, 2019-11-20:
Hello ! Sorry, I forgot to answer you but thanks a lot, it is now working for me ! But, when I try to select more than 3 tags, I get the same error as before. Here is my code : 
`reader.set_read_plan([1],'GEN2', bank=['epc'], epc_target = [b'201909180000000000000006', b'201909180000000000000008', b'201909180000000000000002', b'201909180000000000000010'])
reader.read()`
And I get the error :  `RuntimeError : Unsupported operation` 

Thank you in advance