gotthardp/python-mercuryapi #83: Read Tags list for Gen2 not manipulateable.

Knells1, 2019-08-27:
I am a little newbie in this library, when i tried to used it for my software, i was able to read the tags successfully, when i tried to use the list in a loop, i was not able to compare the tags, rather the comparison statement was being skipped. 

Below is the code snippet:
`#!/usr/bin/env python3
from __future__ import print_function
import time
import mercury
import sys
import os
from subprocess import Popen
reader = mercury.Reader("tmr:///dev/ttyUSB0", baudrate=115200)
t1 = b'E2801170000002094BE1DB86'
tagsav = [t1]
tag = [t1]
reader.set_region("EU3")
reader.set_read_plan([1], "GEN2", read_power=1900)
tags = reader.read()
print(tags)
i = len(tags)
if i ==0:
    #do something
else:
    for x in tags: #new tag list read
        for a in tagsav: #tags saved to compare to
            if a == x:
                for t in tag: #recently saved list to avoid reduntancy
                    if t != a:
                        #do something
                    else:
                        #do something
                if l == 5:
                    #do something
sleep(5)
print("Done")
reader.stop_reading()`

when i scan a tag t1, nothing happens, the program goes to sleep(5) then prints Done.

gotthardp, 2019-08-27:
If you read the same tag twice you get different metadata like rssi,  so two **readings** of the same tag are **not** identical. You can use the *epc* property to find tags that were reread.

Knells1, 2019-08-27:
> If you read the same tag twice you get different metadata like rssi, so two **readings** of the same tag are **not** identical. You can use the _epc_ property to find tags that were reread.

yes i am using the epc, but my issue issue i am not able to compare any element from the list with known tags, maybe the way that i am comparing the strings is not right, it gives me a "mercury.TagReadData, object not subscriptable error"

gotthardp, 2019-08-27:
I don't see any `epc` in your snipplet above. The `a == x` will be never true, because `a` is a bytestring and `x` is `mercury.TagReadData`. You should do `a == x.epc`. Have you tried this?

Knells1, 2019-08-27:
i did not put `epc` because i believe by default reader.read() will give `epc` as output if read data is not specified, and yes i am getting `epc` when i print the output.

i will try `x.epc` and see the result, thanks

Knells1, 2019-08-27:
@gotthardp `x.epc` killed it. thanks a lot.

closed 2019-08-27 by Knells1

gotthardp, 2019-08-27:
Yeah, it's a little confusing. If you print `mercury.TagReadData` it **prints** the EPC, but you still have the object.