gotthardp/python-mercuryapi #85: Issue: Read error when Read Plan is set, Cannot read User or TID memory banks

Chikenbagelz, 2019-09-03:
Hello!
I am currently getting a runtime error after setting the read plan for the ThingMagic M6 reader. I combined these two issues since I believe they might be related to one another. My code is below along with the error message for reading after setting the read plan:

reader = mercury.Reader(uri_string)
reader.set_read_plan(reader.get_antennas(), "GEN2", bank = "user")
tags_read = reader.read(500) <- Line with error

RuntimeError: Invalid Argument

This error only occurs when I have set the read plan, otherwise the read function works fine for me.

When I try to read from any memory bank, even using your example code, I get an Unsupported Operation Error:

reader.set_read_plan(reader.get(antennas), "GEN2", bank = "user")
tags_read = reader.read_tag_mem(1, 0x08, 8)

and without the set_read_plan I still get the same error.

Any ideas?

gotthardp, 2019-09-04:
Hello. Both functions work for me. In general, `RuntimeError` indicates an error from the reader, so you probably have a reader or a tag that does not support the argument/operation you were trying to make. For example, the memory operation (the size, the bank or the offset) may not be supported by the reader you have.

Chikenbagelz, 2019-09-04:
Hello Petr! So I tested writing to the USER memory of the tags and the EPC using the Universal Reader Assistant and was able to just fine. When I then hooked the reader back up to my raspberry pi and ran the code, I was able to write to the EPC of the tag. However, when I powered down the reader and then restarted it and ran the same code I am no longer able to write to the tag's EPC. It is my best guess that there is some issue with letting know the reader which protocol it is using to write to the tags (GEN2). Is there any code that you have before trying to write to the tags (both user and EPC)? Thank you for all of your help!

gotthardp, 2019-09-05:
Alright, I found something: operations not using the read-plan (all write operations and tag_mem operations) depend on a protocol setting that was not configurable. This may be your issue.

I made an update and the `Reader` constructor now accepts *protocol* parameter that allows you to set a protocol to be used for operations when a read-plan is not used.

Chikenbagelz, 2019-09-05:
Hello Petr, I am looking into the new Reader constructor now! I may be completely wrong, but it seems that I get a syntax error (postional arguments follows keyword argument) when I use the new constructor. Would you happen to have an example that works for you? Thank you for all of your help!

gotthardp, 2019-09-05:
For example,
```python
mercury.Reader(uri, baudrate=115200, protocol="GEN2")
```

Chikenbagelz, 2019-09-05:
Hello Petr! I have tried a few different ways and all of them seem to have some error after running the makefile again. 

When I have the same syntax as before:
mercury.Reader(uri)
I am able to connect to the reader fine, but still cannot write or use other memory blocks besides the EPC.

When I have:
mercury.Reader(uri, baudrate=115200)
I get a TypeError: Key not found.

When I have:
mercury.Reader(uri, baudrate=115200, protocol = "GEN2")
I get TypeError: function takes at most 2 arguments (3 given)

I might have made some mistake along the way, but was wondering if you had any issues with the build or was wondering how you updated it for yourself? I can imagine I messed up while running the makefile to update the code.

Thank you!

gotthardp, 2019-09-06:
Negative, the following works for me:
```python
reader = mercury.Reader("tmr:///dev/ttyUSB0", baudrate=115200, protocol = "GEN2")
```
I suspect you have two versions of the module installed, or you installed a wrong version, or you forgot to build it... something with build and installation must be wrong.

gotthardp, 2019-09-06:
I always do `make` and then `sudo make install`. That's it.

Once I had an issue with two different versions installed, when I installed one version, but python used another version. The safest is to make sure there is only one `mercury` module. The trouble is that on Linux it may be installed in many different directories:
```
/usr/lib/python3/dist-packages
/usr/lib/python3.?/dist-packages
/usr/local/lib/python3.?/site-packages
/usr/local/lib/python3.?/dist-packages
```
and similar. I recommend you walk over all the possible directories, make sure there is no mercury in it and then you build and install the module once again.

Chikenbagelz, 2019-09-06:
Hello Petr!

That definitely seemed to be the issue! I removed all copies and rebuilt and installed the module again and the creation of the reader object works great now with the added protocol. I am still receiving an Unsupported operation error whenever I write to my ALN-Pearl (Higgs EC) tags. I believe there is probably some issue in my syntax or with the tags, as I cannot write to the USER memory using the corresponding application from JADAK either (and have already emailed them about this issue). Have you ever encountered this problem?
I appreciate all of your help.

Thank you!

gotthardp, 2019-09-09:
Hello. Well, not all tags support all operations. The tag might be refusing the operation you are executing. I have encountered this when reading too much from a tag memory.

Chikenbagelz, 2019-09-09:
That was exactly my thought. I was able however to do all operations using the Universal Reader Assistant, so it does not seem to be an issue with the tag.

Chikenbagelz, 2019-09-09:
To see if I can even read the USER memory when I know there is something in the memory bank I used (I am only reading one tag):

reader = mercury.Reader(uri, protocol = "GEN2")
tags_read = reader.read(1000)
print("USER:", tags_read[0].user_mem_data)

Each time I receive None as the user_mem_data.

I thought it might be because I did not set the read plan, however when I set the read plan my read command does not function:

reader = mercury.Reader(uri, protocol = "GEN2")
reader.set_read_plan([1], "GEN2" bank = ["user", "epc"])
reader.read(1000)

Error:
RuntimeError: Invalid argument for the reader.read(1000) line.

I tested all of these cases with and without protocol = "GEN2" and just having the "user" specified in the desired bank read.

gotthardp, 2019-09-10:
It is hard for me to debug this. The antenna number could also be a problem. Have you tried this?
```python
reader = mercury.Reader(uri, protocol = "GEN2")
reader.set_read_plan(reader.get(antennas), "GEN2" bank = ["user", "epc"])
reader.read(1000)
```

Chikenbagelz, 2019-09-10:
Hello Petr!

I tried using reader.get_antennas() and I still get the RuntimeError: Invalid argument error for the reader.read(1000) line.

Chikenbagelz, 2019-09-19:
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mercury
>>> reader = mercury.Reader("tmr://169.254.31.226", protocol = "GEN2")
>>> reader.read(1000)
[EPC(b'E200420DA2506014009ECDF9')]
>>> tags_read = reader.read(1000)
>>> print(tags_read[0].user_mem_data)
None
>>> print(tags_read[0].epc_mem_data)
None
>>> reader.set_read_plan([1], protocol = "GEN2", bank = ["user", "epc"])
>>> reader.read(1000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Invalid argument
>>>

Chikenbagelz, 2019-09-24:
Hello!
I got a new reader, the USB Plus, and am now able to read and write the tag's USER memory using the read/write_tag_mem. However, I am still unable to read the epc and user memory banks in one read using set_read_plan. Any help is appreciated, I am not getting any runtime errors or errors at all from the code:

reader = mercury.Reader("tmr:///dev/ttyUSB0", protocol = "GEN2")
reader.set_read_plan(reader.get_antennas(), protocol = "GEN2", bank = ["user", "epc"])
tags_read = reader.read(1000)
tags_read[0].user_mem_data # Just looking at our first tag read (all tags read have non null user memory)

The last line just returns none, even though with read_tag_mem I am able to read what is in the USER memory bank. Is there an issue with how I am trying to access the memory? Thank you for all of your help.

kappler333, 2019-09-25:
Hello,

I have been following this issue and very much interested in the solution.  I am unable to read *both* EPC and USER dataa the same time.  Is the following code expected to return "None" for both banks (tag.epc_mem_data and tag.user_mem_data)?  Or, can you provide the Python code to do so? 
 Please advise.

From San Francisco,
Michael

>>> CODE
#!/usr/bin/python3
import mercury
reader = mercury.Reader("tmr:///dev/ttyUSB0")
print ("MODEL:",reader.get_model())
print ("SERIAL:",reader.get_serial())
reader.set_read_plan(reader.get_antennas(),protocol="GEN2",bank=["epc","user"])
tags = reader.read()
print ("READ: ",tags)
for tag in tags:
	print ("EPC:",tag.epc)
	print ("EPC MEM:",tag.epc_mem_data)
	print ("USR MEM:",tag.user_mem_data)
print ("MEM1:",reader.read_tag_mem(1,0x00,16))
print ("MEM2:",reader.read_tag_mem(2,0x00,16))
print ("MEM3:",reader.read_tag_mem(3,0x00,16))

>>> OUTPUT
MODEL: M5e Compact
SERIAL: 331704701240
READ:  [EPC(b'037019000000000000000181')]
EPC: b'037019000000000000000181'
EPC MEM: None
USR MEM: None
MEM1: bytearray(b'M\x054\x00\x03p\x19\x00\x00\x00\x00\x00\x00\x00\x01\x81')
MEM2: bytearray(b'\xe2\x80\x11\x05 \x00p\xc2\xfeo\t\x91\x00\x00\x00\x00')
MEM3: bytearray(b'u=jho\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')

Chikenbagelz, 2019-12-06:
Problems running the USER Mem read on the IZAR and USB Plus Reader, but works perfectly fine on the USB Pro and Micro readers.