gotthardp/python-mercuryapi #86: Unable to install library using pip

Xzib, 2019-09-29:
While installing the library in windows, it is giving the following error 

mercury.c:23:10: fatal error: tm_reader.h: No such file or directory
     #include <tm_reader.h>

gotthardp, 2019-10-07:
It feels you didn't download the Mercury API. Try to folllow the build instructions in README.

kenghzou95, 2019-11-07:
+1 unable to install the library using pip 
OS: ubuntu 18

gotthardp, 2019-11-07:
The error also appears when `xsltproc ` is not installed. Please make sure you have all software required to build it (see the documentation):
```bash
sudo apt-get install unzip patch xsltproc gcc libreadline-dev python-dev python-setuptools
```

closed 2019-12-03 by gotthardp

clickworkorange, 2020-06-26:
Another trap for young players: the installation will throw this error also if it runs out of space in `/tmp`. I know the space requirement is mentioned in the installation instructions (very helpful, thank you!), but not being one to read such things too carefully (I know!) I was initially stumped by `fatal error: tm_reader.h: No such file or directory`, and it took me a moment to find the true reason for it. The `/tmp` in my case is a  `tmpfs` partition on a Pi3B, so nowhere near the required size. Installing with `pip3 install python-mercuryapi -b /var/tmp` got around the error (with `/var/tmp` being on <strike>disk</strike> microSD).

ahmedmibrahim, 2023-02-20:
pip3 install python-mercuryapi -b /var/tmp still fails as well

pi@raspberrypi:~/python-mercuryapi $ pip3 install python-mercuryapi -b /var/tmp
DEPRECATION: The -b/--build/--build-dir/--build-directory option is deprecated and has no effect anymore. pip 21.1 will remove support for this functionality. A possible replacement is use the TMPDIR/TEMP/TMP environment variable, possibly combined with --no-clean. You can find discussion regarding this at https://github.com/pypa/pip/issues/8333.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting python-mercuryapi
  Using cached python-mercuryapi-0.5.3.tar.gz (21 kB)
Building wheels for collected packages: python-mercuryapi
  Building wheel for python-mercuryapi (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-o5n00f6t/python-mercuryapi_6fcf76df24594a4eb2065aad8ac2c0ba/setup.py'"'"'; __file__='"'"'/tmp/pip-install-o5n00f6t/python-mercuryapi_6fcf76df24594a4eb2065aad8ac2c0ba/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-01ic_h8i
       cwd: /tmp/pip-install-o5n00f6t/python-mercuryapi_6fcf76df24594a4eb2065aad8ac2c0ba/
  Complete output (25 lines):
  running bdist_wheel
  running build
  curl https://www.jadaktech.com/wp-content/uploads/2019/10/mercuryapi-1.31.1.36-2.zip -o mercuryapi-1.31.1.36-2.zip
    % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                   Dload  Upload   Total   Spent    Left  Speed
  100 88113    0 88113    0     0  49252      0 --:--:--  0:00:01 --:--:-- 49225
  unzip mercuryapi-1.31.1.36-2.zip
  Archive:  mercuryapi-1.31.1.36-2.zip
    End-of-central-directory signature not found.  Either this file is not
    a zipfile, or it constitutes one disk of a multi-part archive.  In the
    latter case the central directory and zipfile comment will be found on
    the last disk(s) of this archive.
  unzip:  cannot find zipfile directory in one of mercuryapi-1.31.1.36-2.zip or
          mercuryapi-1.31.1.36-2.zip.zip, and cannot find mercuryapi-1.31.1.36-2.zip.ZIP, period.
  make: *** [Makefile:23: mercuryapi-1.31.1.36/.done] Error 9
  running build_ext
  building 'mercury' extension
  creating build
  creating build/temp.linux-armv7l-cpython-39
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -Ibuild/mercuryapi/include -I/usr/include/python3.9 -c mercury.c -o build/temp.linux-armv7l-cpython-39/mercury.o
  mercury.c:23:10: fatal error: tm_reader.h: No such file or directory
     23 | #include <tm_reader.h>
        |          ^~~~~~~~~~~~~
  compilation terminated.
  error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for python-mercuryapi
  Running setup.py clean for python-mercuryapi
Failed to build python-mercuryapi
Installing collected packages: python-mercuryapi
    Running setup.py install for python-mercuryapi ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-o5n00f6t/python-mercuryapi_6fcf76df24594a4eb2065aad8ac2c0ba/setup.py'"'"'; __file__='"'"'/tmp/pip-install-o5n00f6t/python-mercuryapi_6fcf76df24594a4eb2065aad8ac2c0ba/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-07ci9wp2/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/pi/.local/include/python3.9/python-mercuryapi
         cwd: /tmp/pip-install-o5n00f6t/python-mercuryapi_6fcf76df24594a4eb2065aad8ac2c0ba/
    Complete output (23 lines):
    running install
    /home/pi/.local/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
      warnings.warn(
    running build
    unzip mercuryapi-1.31.1.36-2.zip
    Archive:  mercuryapi-1.31.1.36-2.zip
      End-of-central-directory signature not found.  Either this file is not
      a zipfile, or it constitutes one disk of a multi-part archive.  In the
      latter case the central directory and zipfile comment will be found on
      the last disk(s) of this archive.
    unzip:  cannot find zipfile directory in one of mercuryapi-1.31.1.36-2.zip or
            mercuryapi-1.31.1.36-2.zip.zip, and cannot find mercuryapi-1.31.1.36-2.zip.ZIP, period.
    make: *** [Makefile:23: mercuryapi-1.31.1.36/.done] Error 9
    running build_ext
    building 'mercury' extension
    creating build
    creating build/temp.linux-armv7l-cpython-39
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -Ibuild/mercuryapi/include -I/usr/include/python3.9 -c mercury.c -o build/temp.linux-armv7l-cpython-39/mercury.o
[... cut ...]  __file__='"'"'/tmp/pip-install-o5n00f6t/python-mercuryapi_6fcf76df24594a4eb2065aad8ac2c0ba/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-07ci9wp2/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/pi/.local/include/python3.9/python-mercuryapi Check the logs for full command output.

clickworkorange, 2023-02-22:
Your formatting is _terrible_, but I think pip is trying to unzip a 404 page; https://www.jadaktech.com/wp-content/uploads/2019/10/mercuryapi-1.31.1.36-2.zip does not exist any more, so the installer is curling an 88kb HTML page into a file called mercuryapi-1.31.1.36-2.zip. That won't work. 

As it stands I think installing via pip is a no-go, and you have to build manually instead. Grab the latest release at https://github.com/gotthardp/python-mercuryapi/archive/refs/tags/v0.5.3.tar.gz and untar it to a folder, then download the mercuryapi ZIP file separately and place it in the python-mercuryapi folder. Then edit the Makefile setting for APIZIP to the name of the ZIP-file you downloaded before building.

DoongLi, 2023-03-29:
@clickworkorange , Is there a detailed order? I don't know exactly how to do it

rasdehya, 2024-03-10:
```
10:42:29  |base|root@trixie python-mercuryapi-0.5.3 → make
unzip mercuryapi-1.31.1.36-2.zip
unzip:  cannot find or open mercuryapi-1.31.1.36-2.zip, mercuryapi-1.31.1.36-2.zip.zip or mercuryapi-1.31.1.36-2.zip.ZIP.
make: *** [Makefile:23 : mercuryapi-1.31.1.36/.done] Erreur 9
```
I downloaded the tar and zip version 0.5.3, put the zip in the python-mercury api folder, edit the Make file like so :
```
...
$(APIZIP): ./python-mercuryapi-0.5.3.zip
```
I try also ` $(APIZIP): python-mercuryapi-0.5.3.zip` or `$(APIZIP): /home/$USER/python-mercuryapi-0.5.3.zip ` .
Why it ask for a 1.31.1.36.zip ? Is it a Make file from the future  :-))

clickworkorange, 2024-03-18:
In the folder `python-mercuryapi` (v0.5.4) I have the ZIP file `mercuryapi-AHAB-1.35.2.72-1.zip`. In the `Makefile` I have 
```
APIZIP ?= mercuryapi-AHAB-1.35.2.72-1.zip
APIVER ?= 1.35.2.72
```
I open a terminal, `cd` to `python-mercuryapi` and type `make`. This builds the project without errors. A few suggestions that may help: 

- Get the latest version of `python-mercuryapi` (v0.5.4) by typing `git clone https://github.com/gotthardp/python-mercuryapi`. You shouldn't need to make any changes to the `Makefile`.
- Download `mercuryapi-AHAB-1.35.2.72-1.zip` to the folder created by the previous command. You can get this from https://www.jadaktech.com/documents-downloads/thingmagic-mercury-api-firmware-v1-35-2 (registration required). 
- If you want to use an earlier version of Jadak's API file, edit _only_ the first two lines in the `Makefile` to match your filename and version. For example: 
```
APIZIP ?= mercuryapi-1.31.4.35.zip
APIVER ?= 1.31.4.35
```
I have tested and confirmed that v1.31.4.35 also builds correctly with the latest version of `python-mercuryapi` (v0.5.4). There is also a newer version available from Jadak, v1.37.2
https://www.jadaktech.com/documents-downloads/thingmagic-mercury-api-1-37-2/ which may also work, though I have not tested this. HTH!

lefty01, 2024-03-18:
in #142 I mentioned that I'm successfully using api 1.37.2  (also check this PR #148 )
maybe you could try that version as well ... ?

rasdehya, 2024-03-18:
ok, that was a dumb thing for me in fact,.. i was using the code and the zip from the realease page. 
If i clone the repo  everything works out the box.

So the 'latest' was not really the 'LATEST'.

So if  it helps. just clone the repo. Modification has been made there

sanjaygorak, 2024-11-01:
my patch file opens after the make and it doesnt end and the program gets terminated any help on this!!

clickworkorange, 2024-11-01:
> last lines after the build. patch -p0 -d mercuryapi-1.37.4.8 < mercuryapi.patch ^CTerminate batch job (Y/N)? Y make: *** [Makefile:24: mercuryapi-1.37.4.8/.done] Error 255

I really have no idea; I haven't used Windows in more than a decade. But "Error 255" suggests that `make` couldn't find a command, or that the command failed to run. I suspect this might be relevant: 

https://docs.ros.org/en/galactic/How-To-Guides/Installation-Troubleshooting.html#patch-exe-opens-a-new-command-window-and-asks-for-administrator

What happens if you try to run just `patch` from a cmd prompt?

sanjaygorak, 2024-11-08:
Hey @clickworkorange thank you for responding, and yaa the error 255 when i check it with verbose i get this : 

make VERBOSE=1
make -C mercuryapi-1.35.0.82/c/src/api
make[1]: Entering directory 'C:/Users/sanju/python-mercuryapi/mercuryapi-1.35.0.82/c/src/api'
SOURCE_DIR=lib PATCH_DIR=lib XML_DIR=lib sh lib/install_LTKC.sh NULL lib
'SOURCE_DIR' is not recognized as an internal or external command,
operable program or batch file.
make[1]: *** [Makefile:176: lib/LTK] Error 1
make[1]: Leaving directory 'C:/Users/sanju/python-mercuryapi/mercuryapi-1.35.0.82/c/src/api'
make: *** [Makefile:14: mercuryapi] Error 2

These are the errors i am getting, and if i run patch from cmd prompt it just opens a window of patch.exe and it is just blank. 
If you can help me with something, that would be great, Thank you for your response.

sanjaygorak, 2024-11-15:
@gotthardp can you help me with this