gotthardp/python-mercuryapi #92: Unable to get_temperature() while start_reading()

HanYangZhao, 2019-11-05:
`Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "./thingMagicRead2.py", line 48, in heartbeat
    temp = reader.get_temperature()
RuntimeError: Unimplemented feature
`

I'm assuming this is an error returned from the MercuryAPI?

gotthardp, 2019-11-05:
Yes. It is likely not supported by your reader, or the reader is in some state in which it is not allowed to call that function. I cannot do much about that.

natcl, 2019-11-05:
Seems to be a duplicate of #71

HanYangZhao, 2019-11-05:
I dug a bit further with the URA assistant by printing the serial log and was able to see the temperature being sent by the reader. So I went to check the doc and found the following 
[image]

gotthardp, 2019-11-05:
Does this suggest something?

HanYangZhao, 2019-11-05:
[image]

From what I understand, the temperature is collected by the statusListener, It seems to be similar to the ReadListener used for tag reading. I was however unable to find the format of the data that's being passed(something similar to TMR_TagReadData).

gotthardp, 2019-11-06:
Yeah, it seems that TMR_addStatsListener could be used, but it is not implemented right now. Contributions are welcome :)

HanYangZhao, 2019-11-07:
I have attempted to add the TMR_addStatsListener on my [fork](https://github.com/HanYangZhao/python-mercuryapi). So far it doesn't return any error, but the callback doesn't seem to get called. Maybe you have some insights?

gotthardp, 2019-11-08:
Unfortunatelly not.

gotthardp, 2019-11-11:
This was fixed by @HanYangZhao in https://github.com/gotthardp/python-mercuryapi/commit/7f493198056824cba9f8b11e89eadf46c4720de2. Function `enable_stats` may now e used to retrieve temp (and other stats) during reading. Thanks a lot again!

closed 2019-11-11 by gotthardp